UnchartedBull/OctoDash #1884: Octodash not found / Xinit: connection to server lost

N3orun, 2021-05-30:
Hi, 

I successfully installed, verified each step 3 times again, tried the troubleshooting hints. Still stuck after reboot. 

Raspi Pi 3 modell: [https://philipp.d.pr/QaGLbF/IHGowhOAUd](url)

Error Screenshot. [https://philipp.d.pr/rv9ZLh/sh82fvvKjl](url)

Using with a BIGTREETECH PI TFT50 Touch Display

HDMI is only showing a rainbow after boot. 
 
Any Ideas?

N3orun, 2021-05-31:
[image]

UnchartedBull, 2021-06-01:
It says that `octodash` isn't installed. Please manually install the `.deb` file from the release page with `dpkg`. Last two steps from here: https://github.com/UnchartedBull/OctoDash/wiki/Installation#manual-installation

N3orun, 2021-06-01:
seems like updating octopi to the latest version before running the script caused an issue. ran again on a fresh sd and without configuring or updating octopi before and it seems to boot at the moment. Thanks anyway! 

Keep up the great work! Danke!

closed 2021-06-17 by UnchartedBull

EmEiBee, 2023-08-26:
Problem is back:
Same here with the actual stuff from today on PI4b-4gb