UnchartedBull/OctoDash #1888: [ENH] File search and default SD or Local

Qualith, 2021-06-01:
When you have a lot of files and want to print one in particular, it would be very useful if you can search by file name.
I have to go to the PC to do it via web.

Another useful feature would be to disable SD (or local) or just select a default option by default. For expample, I don't use the SD option, so I can select file menu and it should enter Local by default.

Thank you for your excellent work.

UnchartedBull, 2021-06-01:
Search probably will be not ideal on most touchscreens, plus the virtual keyboard feature isn't up and running yet. You can maybe try sorting by upload time, to get the newest files first.

You can disable the SD option by either un-mounting the SD in OctoPrint or just physically ejecting the SD card from the printer. That should hide the menu point.

Hillshum, 2025-03-21:
The ability to select a default folder was added in v2.4.0.