UnchartedBull/OctoDash #1892: NOX theme V0.2.2 out of range for filament change screen

Jack77777777, 2021-06-02:
@noxhirsch can you adjust this ? See attached picture.
[image]

noxhirsch, 2021-06-02:
I'll try to fix it later today or tomorrow 👍

noxhirsch, 2021-06-03:
After some trial & error without succes, I checked if this also happens without the NOX theme - and it does indeed.
The way I see it, two changes need to be made in OctoDash, but I have no time to setup a dev instance at the moment, to try that out. In theory, this should fix the problem:

Either the second `<span>` must be integrated into the first one - or both spans need a wrapper div/span:
https://github.com/UnchartedBull/OctoDash/blob/2ed9b86473a6451150a2fe0004a6962d1929061d/src/app/filament/change-filament/change-filament.component.html#L17-L20

I'd prefer the first solution, if it doesn't cause any problems. But if a wrapper around both spans is used, these styles should be moved to the wrapper:
https://github.com/UnchartedBull/OctoDash/blob/2ed9b86473a6451150a2fe0004a6962d1929061d/src/app/filament/change-filament/change-filament.component.scss#L18-L23
Then either the wrapper or the two spans need to have `display:inline;` instead of `display:block;` (I'm not sure if both works, I would have to try that out)

UnchartedBull, 2021-06-03:
Oh this must've happened with the i18n integration. I'm going to fix this. Thanks for reporting / providing a fix!

Jack77777777, 2021-06-05:
Is this already fixed ? If yes, what do i have to do ?

UnchartedBull, 2021-06-07:
Not fixed yet. Will be released with the next version.

PR #1997 (2021-06-26): fix filament screens misalignment
fixes #1892
closed 2021-06-26 by UnchartedBull