UnchartedBull/OctoDash #1910: Display options

tfaz099, 2021-06-04:
Looking to see if you can add some options to the display (specifically in printing mode):
1-Time of Day Clock
2-Current Z-Offset Setting
3-Dim Display Button to put display to sleep when you want to start a prin and go to bed :P
4-And when in Paused Mode a function to Load and Unload Filament with a configurable Step/Length 

Thank you!

UnchartedBull, 2021-06-07:
Most of the stuff is already implemented (or already worked on):

2 - will be shown in Custom Actions
3 - just disable DLP and you'll get Z-Height instead of Layers
4 - other issue open for that already
5 - you can access the filament change routine from the pause screen

tfaz099, 2021-06-08:
I was hoping on #3 for both on the same screen... all good, and also, note that there is no move E on the control menu, only X and Y

UnchartedBull, 2021-06-29:
There is a PR open for E-Movement in the Control screen, so that will come in a future version. For everything else there are already separate issues open, so I'm closing this here. It won't be possible to display Z-Height and Layer Height on the same screen though, as that will just be too crowded. Feel free to create a custom version for yourself.

closed 2021-06-29 by UnchartedBull