UnchartedBull/OctoDash #1918: Turn on PSU on wake

dxgldotorg, 2021-06-05:
**Is your feature request related to a problem? Please describe.**
When waking OctoDash the PSU Control plugin should be invoked if the printer's power is off.

**Describe the solution you'd like**
Turn on 3D printer power when waking the OctoDash.

**Link other projects**
Uses PSU Control - https://plugins.octoprint.org/plugins/psucontrol/

**Additional context**
Currently I have to use my phone or my PC to power on the 3D printer; this would help make it easier to power on.

jneilliii, 2021-06-05:
There is a setting for this to power on when exiting sleep. That's how I use it. You can also use custom actions ans have a button.

dxgldotorg, 2021-06-05:
Thanks for the reply.  Could you point me in the right direction of that setting?

jneilliii, 2021-06-05:
in the UI it's turn on printer when exiting sleep, under OctoDash > General.

closed 2021-06-05 by dxgldotorg