UnchartedBull/OctoDash #1919: Build Dev in Windows

jneilliii, 2021-06-07:
**What doesn't work?**
Trying to npm run build latest main branch as of the time of this post and getting error.

```
App threw an error during load
Error: reference "http://example.com/root.json#/properties/plugins/properties/printTimeGenius/properties/enabled" resolves to more than one schema
    at ambiguos (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\ajv\dist\compile\resolve.js:149:16)
    at Ajv.addRef (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\ajv\dist\compile\resolve.js:116:23)
    at C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\ajv\dist\compile\resolve.js:109:29
    at _traverse (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\json-schema-traverse\index.js:69:5)
    at _traverse (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\json-schema-traverse\index.js:80:13)
    at _traverse (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\json-schema-traverse\index.js:80:13)
    at _traverse (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\json-schema-traverse\index.js:80:13)
    at module.exports (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\json-schema-traverse\index.js:14:3)
    at Ajv.getSchemaRefs (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\ajv\dist\compile\resolve.js:103:5)
    at Ajv._addSchema (C:\Users\jneil\Documents\3D Prints\PrinterUprades\OrdBotHadron\OctoprintRelated\OctoDash\node_modules\ajv\dist\core.js:442:51)
```

**What did you already try?**
Tried `npm clean-install` and build again and same result.

**General Information:**
 - ASUS Laptop i7 processor
 - Windows 10, node v14.15.4, npm v7.16.0
 - OctoDash Version v2.2.0
 - OctoPrint Version v1.6.1 (n/a)

**Additional context**
Just the error listed above. Any hints would be greatly appreciated.

UnchartedBull, 2021-06-09:
Hmmm this is really weird and shouldn't happen. I only run macOS und Linux, so I can't reproduce this on my machine, but my best guess would be to change the `ajv version` up (if available) or down and check whether that solves the issue.

UnchartedBull, 2021-06-09:
Ok I was just able to replicate this on my Linux machine. Going to push a fix to master directly after fixing it!

UnchartedBull, 2021-06-09:
This should be fixed with https://github.com/UnchartedBull/OctoDash/commit/036561e2c8e2111091017b5d7aa2c466d16868f2. Please reopen if the issue reappears / doesn't get fixed :)

closed 2021-06-09 by UnchartedBull