UnchartedBull/OctoDash #1923: Webcam preview showing scroll bar

fhb, 2021-06-07:
Hi, I hope it's appropriate to post this here: 

I just tried adding the Camera with the latest Octodash release 2.2.0 with and without the Multicam plugin (BTW the [wiki entry ](https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions) doesn't mention the !WEBCAM code jet):
`[!WEB]http://localhost/plugin/octodashcompanion/webcam?webcam=1`
`[!WEB]http://localhost/plugin/octodashcompanion/webcam`
`[!WEB]http://localhost/webcam/?action=stream`

Camera is a C270 with 4:3 aspect ratio, the display is a Hyperpixel 4.0. 

> scrolling is unexpected for sure. do you have the muticam plugin installed? 

I always get the same result with scroll bars on the right side. This should not be the case, right?
[image]

> Just released the multicam support version 0.0.5.

Do I need to update to a beta release or the most recent GitHub version? How do I get Multicam support version 0.0.5?

jneilliii, 2021-06-07:
The [!WEBCAM] option is built into my plugin, not core OctoDash. It's just a way for me to identify the selection and replace it with the URL for the webcam view page provided by the plugin using the `[!WEB]` option that is available in OctoDash. Scroll bars should not exist when using those, like the below, but make sure that you have restarted OctoDash after making the changes within the plugin.

If you don't have multicam plugin installed then the following should be all you need.

```
[!WEB]http://localhost/plugin/octodashcompanion/webcam
```

fhb, 2021-06-07:
Thank you for clarifying the difference between Octodashcompanion and Octodash!

Octodash, Octoprint and the RPi running it have been restarted several times after making changes to the custom action already. 
```
[!WEB]http://localhost/plugin/octodashcompanion/webcam?webcam=1
[!WEB]http://localhost/plugin/octodashcompanion/webcam
```
Both still result in the iframe with the scroll bars. The camera feed seems to be very few pixels too high. How can I provide more debugging information?

jneilliii, 2021-06-07:
can you share your webcam stream settings from OctoPrint and what resolution are you using in octopi.txt?

fhb, 2021-06-09:
Sure, it's pretty much be standard configuration:

**Screenshot**
[image]

**/boot/octopi.txt**

```
### Windows users: To edit this file use Notepad++, VSCode, Atom or SublimeText.
### Do not use Notepad or WordPad.

### MacOSX users: If you use Textedit to edit this file make sure to use 
### "plain text format" and "disable smart quotes" in "Textedit > Preferences"

### Configure which camera to use
#
# Available options are:
# - auto: tries first usb webcam, if that's not available tries raspi cam
# - usb: only tries usb webcam
# - raspi: only tries raspi cam
#
# Defaults to auto
#
#camera="auto"

### Additional options to supply to MJPG Streamer for the USB camera
#
# See https://faq.octoprint.org/mjpg-streamer-config for available options
#
# Defaults to a resolution of 640x480 px and a framerate of 10 fps
#
#camera_usb_options="-r 640x480 -f 10"

### Additional webcam devices known to cause problems with -f
#
# Apparently there a some devices out there that with the current 
# mjpg_streamer release do not support the -f parameter (for specifying 
# the capturing framerate) and will just refuse to output an image if it 
# is supplied.
#
# The webcam daemon will detect those devices by their USB Vendor and Product
# ID and remove the -f parameter from the options provided to mjpg_streamer.
#
# By default, this is done for the following devices:
#   Logitech C170 (046d:082b)
#   GEMBIRD (1908:2310)
#   Genius F100 (0458:708c)
#   Cubeternet GL-UPC822 UVC WebCam (1e4e:0102)
#
# Using the following option it is possible to add additional devices. If
# your webcam happens to show above symptoms, try determining your cam's
# vendor and product id via lsusb, activating the line below by removing # and 
# adding it, e.g. for two broken cameras "aabb:ccdd" and "aabb:eeff"
#
#   additional_brokenfps_usb_devices=("aabb:ccdd" "aabb:eeff")
#
# If this fixes your problem, please report it back so we can include the device
# out of the box: https://github.com/guysoft/OctoPi/issues
#
#additional_brokenfps_usb_devices=()

### Additional options to supply to MJPG Streamer for the RasPi Cam
#
# See https://faq.octoprint.org/mjpg-streamer-config for available options.
#
# NOTE: Newer raspi cam modules are reporting as usb devices causing these
#       options to be ignored. Set `camera="raspi"` to avoid these issues.
#
# Defaults to 10fps
#
#camera_raspi_options="-fps 10"

### Configuration of camera HTTP output
#
# Usually you should NOT need to change this at all! Only touch if you
# know what you are doing and what the parameters mean.
#
# Below settings are used in the mjpg-streamer call like this:
#
#   -o "output_http.so -w $camera_http_webroot $camera_http_options"
#
# Current working directory is the mjpg-streamer base directory.
#
#camera_http_webroot="./www-octopi"
#camera_http_options="-n"

### EXPERIMENTAL
# Support for different streamer types.
#
# Available options:
#   mjpeg [default] - stable MJPG-streamer
#   hls - experimental FFMPEG HLS streamer
#camera_streamer=mjpeg```

jneilliii, 2021-06-09:
ok, thanks. 4:3 aspect ratio is something I haven't tested with. All my cameras are setup for 16:9 ratio. You octopi.txt is default so you are running 640x480 by default. I'll do some additional testing on my end and see what's going on.

hawkeyexp, 2021-06-09:
@jneilliii Same as in my case if you remeber  :-) - my cam is 4:3, too.

jneilliii, 2021-06-09:
yeah, probably just needs some css tweaks for height versus width. I have to go back and see what MMcLure did for those.

closed 2021-06-26 by stale[bot]