UnchartedBull/OctoDash #1924: POWERTOGGLE only Powering off

fhb, 2021-06-07:
**Describe the bug**
The !POWERTOGGLE action only turns off my Prusa mini plugged into a TP-Link power plug. !POWERON custom action and automatic power on, when octodash leaves sleep state do work.

**To Reproduce**
Steps to reproduce the behavior:
1. Touch click the [!POWERTOGGLE] custom action
2. Octodash goes to sleep
3. Touch click on the arrow in Octodash to open custom actions
3. Touch click the [!POWERTOGGLE] custom action
4. Nothing happens

**Expected behavior**
The TP-LINK power plug should power on

**Screenshots**
[image]
[image]

**General Information:**
 - Raspberry PI 4
 - OS Info: Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version 2.20
 - OctoPrint Version 1.6.1

jneilliii, 2021-06-07:
This is actually intended because my TPLink Plugin doesn't have a toggle state the same way that PSU Control plugin does. I use a separate button for the on/off commands. If you want the toggle option, you can switch to using PSU Control with the TPLink sub-plugin for it.

fhb, 2021-06-07:
Okay, thank you! It's still unexpected behaviour I think. It's probably a good idea to mention this in the custom actions documentation. I will stick to distinct on/off actions, I actually find it hard to think of a situation where "!POWERON" is necessary thanks to the automation.

closed 2021-06-07 by fhb

7R0J4Ner, 2021-09-20:
I also have this Problem but with a simple Relay, not a Smartplug.

It would be very nice to have just 1 Button to Controll the Power...

Is there a Workaround?

jneilliii, 2021-09-20:
@7R0J4Ner POWERTOGGLE should be supported by the PSUControl plugin with a simple relay.

7R0J4Ner, 2021-09-21:
Well, unfortunately, it does not work. I just tried it again.
Exactly the same error as described by @fhb above.

I have connected the relay to GPIO4 and configured it in PSU Controll with the GPIO plugin.
The commands [!POWERON] + [!POWEROFF] work without problems.
But the command [!POWERTOGGLE] only switches the relay off, but not on anymore.
With [!POWERON] after [!POWERTOGGLE] the relay can be switched on again.

jneilliii, 2021-09-21:
@kantlivelong ?

kantlivelong, 2021-09-21:
https://github.com/kantlivelong/OctoPrint-PSUControl/wiki/Troubleshooting#how-to-provide-logs

7R0J4Ner, 2021-09-24:
First of all, thank you guys!

Here are the Logs.
[octoprint-logs.zip](https://github.com/UnchartedBull/OctoDash/files/7223145/octoprint-logs.zip)

What i did:
07.42.30 -	[!POWERTOGGLE] 	to turn on Printer   -Did work
07.42.52 -	[!POWERTOGGLE] 	to turn off Printer				-Did not work
07.43.09 -	[!POWERTOGGLE] 	again to turn off Printer		-Did not work
07.43.24 -	[!POWERTOGGLE] 	again to turn off Printer		-Did not work
07.43.38 -	[!POWERON]		to turn on Printer				-Did work
07.43.57 - 	[!POWEROFF]		to turn off Printer				-Did work
07.44.17 - 	Tap on Sleepscreen to turn on Printer			-Did work
07.44.43 - 	[!POWERTOGGLE] to turn off Printer				-Did work

All this was done from the Octodash display via Custom Actions.

The status indicator of the relay works correctly in the interface, i.e. the flash turns green when the printer is on and gray when off.

PSU Controll Setup:
[image]

RPI-GPIO Setup:
[image]

Again, Thank you!
If you need anything else, let me know.

kantlivelong, 2021-09-24:
@7R0J4Ner The logs indicate that that PSUControl is receiving the toggle/on/off requests and as you mentioned the UI is updating accordingly. This leaves me to believe you have an issue with your wiring or components. Many users end up purchasing 5V relays not realizing the Pi is 3.3V logic. Some 5V relays will still work but it seems to be a toss in the air. I'd suggest posting your issue on the OctoPrint forum with details about config, components, and wiring.

7R0J4Ner, 2021-09-25:
I know the problem with the relays and have therefore tried several.
I am actually pretty sure that it is not the hardware.
The relay responds without problems when I control it via the commands [!POWERON] & [!POWEROFF].
Also it is addressed with the 1st use of the command [!POWERTOGGLE], only all further executions of [!POWERTOGGLE] have no more effect.

Is it possible to send the command [!POWERTOGGLE] via the Octoprint terminal or ssh? 
Simply using [!POWERTOGGLE] in the Octoprint Terminal did not work.