UnchartedBull/OctoDash #1950: Controlling via gamepad/joystick  or GPIO buttons

kieppie, 2021-06-13:
**feature request related to a problem? describe.**

OD works OK (on RasPi 3B+), but the navigation isn't great on this unit, especially scrolling & navigating.   
Not OD's fault, but directly adversely affect the usability

**Describe the solution**

Would like to request the ability to control LCD interface via HID device, such as a playstation sixaxis controller (wired or BT), [XBox controller](https://github.com/OutsourcedGuru/OctoPrint-Xbox), or just basic keyboard/mouse [input mapping](https://github.com/zeth/inputs).

Even simply having the ability to wire up a [simple joystick via GPIO](https://www.sparkfun.com/products/15168) would go far.

As an added bonus, have these manual [controls to interact](https://plugins.octoprint.org/plugins/usb_keyboard/) with the printer(s)

**Link other projects**

* [OctoPrint-Xbox](https://github.com/OutsourcedGuru/OctoPrint-Xbox)
* [inputs](https://github.com/zeth/inputs).
* [USB keyboard](https://plugins.octoprint.org/plugins/usb_keyboard/)

jneilliii, 2021-06-13:
I think this is possible, at least I know I've seen people use a mouse, but required updating x configurations I think. It was on discord where I saw it.

UnchartedBull, 2021-06-16:
Controlling via mouse is easily possible, you just need to remove the `-- -nocursor` part from your `~/.bashrc`. 

The other input methods are way too much work to implement (at least for me) and out of scope for OctoDash, so I'm closing this here as wont fix, sorry! I'm happy to accept PRs regarding this topic, so feel free to create one :)

closed 2021-06-16 by UnchartedBull

kieppie, 2021-06-16:
thanks @UnchartedBull .

I'll try that & see if I can provide other functional enhancements