UnchartedBull/OctoDash #1955: octodash go to sleep mode while printing

dmitrimedina, 2021-06-16:
**Describe the bug**
i have this error for the third time in this week, the octodash installed on my raspberry pi 3b+ go to sleep while printing and the printer stop printing the model. I been running octodash for a little more than a month and everything was ok until a few days ago that this start happening 

**To Reproduce**

**Expected behavior**
don't go to sleep mode while printing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context about the problem here.

jneilliii, 2021-06-17:
OctoDash will only go to sleep if your printer becomes disconnected. I think your issue may be in your printer connection rather than OctoDash. What does the terminal tab say in OctoPrint when this happens? Make sure to check right away.

dmitrimedina, 2021-06-17:
In the octoprint web its say to reload the webpage like when you are using octoprint web and you turn of the raspberry pi, and in the screen i have installed to the raspberry pi and use the octodash plugin it say octodash is in sleep mode or something like that. And its say to touch the screen to wake up, so i touch the screen and the octodash displays appear normally again

UnchartedBull, 2021-06-26:
Please provide the version information requested in the issue template. This seems like a setup problem to me, since other users aren't experiencing this.

closed 2021-07-13 by stale[bot]