UnchartedBull/OctoDash #1981: Layer Count different on OctoDash versus Octoprint's gcode viewer.

sonjamichelle, 2021-06-25:
**Describe the bug**
I noticed while doing a print, OctoDash will report say layer 3 when Octoprints gcode viewer is reporting layer 5.

Also Octodash will say something like "Layer 5 of 53" and gcode viewer is reporting "Layers with Extrusion: 55"

**Screenshots**

[image]

[image]

**General Information:**
 - Hardware Raspberry Pi 4 2GB
 - OS Info OctoPi from the raspberry pi imager
 - OctoDash Version: installed from pip with url: bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
 - OctoPrint Version 1.6.1

sonjamichelle, 2021-06-25:
Watching the print I noticed that on the sidebar under state, the layer count matched OctoDash but the times were wildly different.

OctoDash reported an elapsed time of 2h 13m and a remaining time of 3h 12m. You can see in the screenshot that those times are very different

[image]

sonjamichelle, 2021-06-25:
Towards the end of the print the layer reporting was off by 2.  OctoDash reported 53 of 53 and Octoprint reported 55 of 55.

I may have been mistaken about the time reporting. A while after my post I noticed the time was relatively spot on.

I'm not a programmer so I have no clue at was is going on, I can only report what I observe to the best of my ability.

sonjamichelle, 2021-06-25:
Doing another print, the time reporting is correct. Not sure what I was seeing earlier. Could be my mistake.

Though the layer reporting is still off by 2.

[image]

On OctoDash the current print is reporting 17 of 58, the State section of OctoPrint's sidebar reports 17 of 58 too. However, the gcode viewer reports 19 of 60.

Not really sure which one is the correct one now. Could be the gcode viewer.

The Pretty GCode plugin also reports the same as OctoDash and the Sidebar, so it may be looking as if it is the gcode viewer panel. That is the panel distributed with octoprint, not a plug -in, correct?

I hope that I'm not spamming too much, I just want to report as much as possible and be helpful.

UnchartedBull, 2021-06-26:
OctoDash uses the Layer Progress Information provided by DisplayLayerProgress and doesn't have any access to the GCode Viewer.  I think all your screenshots point to something funky going on with the layer count in the GCode Viewer. Probably best to check whether that is natively integrated or a preinstalled plugin and then check that with their devs.

Regarding the print times, I don't see too much discrepancy between the times (only like 2 minutes for each). Does the time difference still happen?

sonjamichelle, 2021-06-26:
I agree, was sort of my thinking too as I explored a bit further. I believe the gcode viewer is native to Octoprint. I'll look there..

As far as the time, that may have been my mistake. I was a bit distracted with another project at the time, so I believe the time reporting is working as intended.

UnchartedBull, 2021-06-29:
Ok, closing this here then. Thanks for all the info!

closed 2021-06-29 by UnchartedBull