UnchartedBull/OctoDash #1995: Octodash has gone green!

fizzyade, 2021-06-26:
I've just powered up my Pi this morning, the Pi boots as normal and octodash started normally, at least at first.  I got the correct "Octodash is initialising" and it flipped to the "connected" screen after octoprint connected to the printed, but almost instantly after that, the octodash screen background turned green.

It's now happily running away and is fully functional, just with a green screen.

I'm running on a Raspberry Pi 3 B+ with a hyperpixel display.  Everything is the latest version of the software, Octodash, Octoprint and so on.

The screen is behaving right up until Octodash switches to the connected screen, where it shows the dashboard, I also briefly see that in the correct colours.  The boot-up until that point is all in the correct colours.

[image]

fizzyade, 2021-06-26:
Humn, it seems this may be an issue with the hyperpixel, the printer is busy at the moment so I will take a look when it's done.

UnchartedBull, 2021-06-26:
Yeah this looks like a loose cable / connection or something. Closing this here. If it actually is OctoDash please reopen.

closed 2021-06-26 by UnchartedBull

fizzyade, 2021-06-27:
I created a fresh installation of octopi, installed the hyper pixel driver, and then onto-dash and it's back working now.  

If I switch back to the other SD card then I get the same screen issue.

The issue started when I changed the screen sleep setting in octodash, but I'm unsure of whether it was a coincidence or something else.

sonypsx, 2022-09-21:
I got exactly the same issue, but with a fresh install. RP4 , Bullseye 64 Bit, Octoprint 1.8.3
Took me hours to figure out how to get the rotation correct (thanks to @alison-gravley in https://github.com/UnchartedBull/OctoDash/issues/2944), stumbled over the missing appindicator3-1 and some more issues. But finally got it working. But it is very greenish. However -> the Display (Hyperpixel4) is 100% OK, as before i was able to run the automatic install script from octodash i had a color correct desktop on my display.
Are there any hints how to get rid of this?

fizzyade, 2022-09-22:
> I got exactly the same issue, but with a fresh install. RP4 , Bullseye 64 Bit, Octoprint 1.8.3
>
> Took me hours to figure out how to get the rotation correct (thanks to @alison-gravley in [...]

I wiped and reinstalled as nothing else worked, I have no idea why it did it and it hasn't happens since.

supawiz6991, 2022-09-24:
> I got exactly the same issue, but with a fresh install. RP4 , Bullseye 64 Bit, Octoprint 1.8.3 Took me hours to figure out how to get the rotation correct (thanks to @alison-gravley in #2944), [...]

I'm going to guess that this is a screen specific issue since I'm running the raspberry pi offical 7" touch display and having had this problem under Raspberry pi OS Bullseye on a Rasperry Pi 4. 

Are you loading the driver for that specific display in /boot/config.txt?

If you did go through that then it may be that the driver is not compatible with Bullseye. Here is the github I'm referencing:
https://github.com/pimoroni/hyperpixel4

Also, this issue from their github page in reference to Bullseye may be of some interest: https://github.com/pimoroni/hyperpixel4/issues/177

In this issue others have had problems with display irregularities similar to yours so maybe it will set you on the right route,