UnchartedBull/OctoDash #2003: Prusa MK3S & MMU2S Sensor Status

buzz-b, 2021-06-29:
Hi,

Firstly, thanks for a great app, works great on my Prusa MK3S with a HyperPixel 4.0.

**Is your feature request related to a problem? Please describe.**
This is not a bug or issue with a Dash problem, however the Prusa MK3S has an add on called the MMU2S whic is a multi filament selector allowing up to 5 filaments to be printed into one extruder. It has numerous moving parts and is a little prone to issues in itself. 

**Describe the solution you'd like**
Adding filament sensor icons or numbering to the dash would be a fantastic troubleshooting aid, instead of having to make up PCB's with LED's as some ppl have done.

Other MMU2S info would be good too if any is available from the unit while printing, however this is a lower priority for me.

**Link other projects**
If you could pass this info through to the HASS (Home Assistant) plugin then that would be great and I could automate an alert if issues occurduring the print.

**Additional context**
None that I can think of.

Hopefully you can accommodate this request somehow, hopefully you can see the Prusa Sensors direct from the Printer connection, and no need for any direct Pi connections.

Thanks again.

UnchartedBull, 2021-06-29:
Hi, yeah I've seen the MMU2S, looks quite interesting. It does seem like it isn't supported that well by OctoPrint. Do you get any indication on OctoPrint on what Filament is selected? If that info isn't present it won't be able to be displayed in OctoDash either (unless a lot of work is done on the OctoPrint side).

I've only come across this plugin: https://plugins.octoprint.org/plugins/mmu2filamentselect/, but that doesn't have an API to retrieve data, so it won't be of much use.

Regarding the HA integration: Please open an issue in their respective repository. I don't have any controls over the data that is being send to HA.

buzz-b, 2021-06-30:
Hi, Yeah thanks for that. I guess I was just hoping you/octoPrint had access to the sensor status in the MK3S?, as this has got sensor status on it's display, but you have to go deep through it's horrible menu. So if you/octoprint don't have access to any sensors on that model then this is not possible. I'm not fussed about filament selection as I can see that easily enough, but the sensors are hidden and can be fussy and incorrect.

I will hassle the octoprint plugin guys for the sensor status in octoprint, then come back to this request when I'm more aware the info is there for you to read. I just figured you may have more pull than me  :)

Thanks again...

UnchartedBull, 2021-07-05:
Sure, please reopen if there is more information available :)

closed 2021-07-05 by UnchartedBull