UnchartedBull/OctoDash #2022: Print Percentage incorrect on octodash

SeekaMartin, 2021-07-03:
showing correct both on printer and octoprint but not OctoDash
[image]
[image]

UnchartedBull, 2021-07-16:
You do have Print Time Genius insatlled right? PTG will update the percentage in the UI, the percentage that is being send to OctoDash isn't updated though. I'm currently looking into ways to solve this. Current idea is to calculate the percentage on my own based on time elapsed and time remaining, as this should be the most useful percentage from my pov.

closed 2021-08-02 by stale[bot]

SeekaMartin, 2021-08-02:
> You do have Print Time Genius insatlled right? PTG will update the percentage in the UI, the percentage that is being send to OctoDash isn't updated though. I'm currently looking into ways to solve [...]

I do but the issue is there with it enabled and disabled, your way to calculate seems best