UnchartedBull/OctoDash #2026: memory leak in octodash v2.2.0

mithkr, 2021-07-06:
**Describe the bug**
System runs out of memory after a while also if idle (no print jobs running).
OOM killer kills octodash, as it has leaked the memory.
`[289804.365120] Out of memory: Killed process 509 (octodash) total-vm:817456kB, anon-rss:249080kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:632kB oom_score_adj:0`

**To Reproduce**
Boot Raspberry Pi with auto stared octodash and wait until system runs out of memory.
Monitoring memory usage with top shows memory usage increase of octodash.

**Expected behavior**
System should not run out of memory.

**Screenshots**
```
top - 09:44:16 up 11:16,  2 users,  load average: 0,18, 0,21, 0,15
Tasks: 120 total,   1 running, 119 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1,4 us,  0,8 sy,  0,0 ni, 97,7 id,  0,0 wa,  0,0 hi,  0,1 si,  0,0 st
MiB Mem :    460,7 total,     40,5 free,    203,7 used,    216,6 buff/cache
MiB Swap:    100,0 total,     18,5 free,     81,5 used.    197,8 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                                     
  509 pi        20   0  608560  80112  31016 S   1,0  17,0   8:06.60 octodash                                                                                                                                                                                    
  401 pi        20   0  410112  75380   7972 S   8,6  16,0  59:32.91 octoprint                                                                                                                                                                                   
  575 pi        20   0  335044  74144  41980 S   0,3  15,7   3:19.44 octodash                                                                                                                                                                                    
  556 pi        20   0  209476  31196  28804 S   0,0   6,6   1:04.19 octodash                                                                                                                                                                                    
  591 pi        20   0  215552  22668  16420 S   0,0   4,8   0:03.32 octodash

top - 13:17:04 up 2 days, 14:48,  2 users,  load average: 0,09, 0,07, 0,07
Tasks: 122 total,   1 running, 121 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1,6 us,  1,6 sy,  0,0 ni, 96,6 id,  0,2 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     25,5 free,    355,8 used,     79,4 buff/cache
MiB Swap:    100,0 total,      0,0 free,    100,0 used.     48,8 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                                     
  509 pi        20   0  768304 220252  19344 S   1,7  46,7  45:16.04 octodash                                                                                                                                                                                    
  575 pi        20   0  355008  72464  26512 S   0,3  15,4  17:26.14 octodash                                                                                                                                                                                    
  401 pi        20   0  410276  68804   3940 S   8,3  14,6 329:22.30 octoprint                                                                                                                                                                                   
  556 pi        20   0  210500  16100  13344 S   0,0   3,4   6:00.81 octodash                                                                                                                                                                                    
  591 pi        20   0  215040  10256   9280 S   0,0   2,2   0:07.32 octodash                                                                                                                                                                                    
  512 pi        20   0  156208   5556   5428 S   0,0   1,2   0:00.14 octodash                

[289804.363927] python invoked oom-killer: gfp_mask=0x100cca(GFP_HIGHUSER_MOVABLE), order=0, oom_score_adj=0
[289804.363944] CPU: 2 PID: 491 Comm: python Tainted: G         C        5.10.17-v7+ #1421
[289804.363947] Hardware name: BCM2835
[289804.363951] Backtrace: 
[289804.363975] [<809e477c>] (dump_backtrace) from [<809e4b0c>] (show_stack+0x20/0x24)
[289804.363982]  r7:ffffffff r6:00000000 r5:60000113 r4:80ee5d54
[289804.363990] [<809e4aec>] (show_stack) from [<809e8cb8>] (dump_stack+0xcc/0xf8)
[289804.363997] [<809e8bec>] (dump_stack) from [<809e6fa8>] (dump_header+0x64/0x208)
[289804.364014]  r10:80e05144 r9:00100cca r8:00000000 r7:80c13384 r6:81f6f600 r5:82c22e80
[289804.364019]  r4:89579d50 r3:df8c3ffb
[289804.364029] [<809e6f44>] (dump_header) from [<802b18b0>] (oom_kill_process+0x1b4/0x1c0)
[289804.364035]  r7:80c13384 r6:89579d50 r5:82c23400 r4:82c22e80
[289804.364043] [<802b16fc>] (oom_kill_process) from [<802b2428>] (out_of_memory+0x2b8/0x390)
[289804.364049]  r7:80e083c0 r6:80e05008 r5:82c22e80 r4:89579d50
[289804.364058] [<802b2170>] (out_of_memory) from [<803018dc>] (__alloc_pages_nodemask+0x7ec/0x1184)
[289804.364064]  r7:80e05784 r6:00000000 r5:ffffe000 r4:00400040
[289804.364072] [<803010f0>] (__alloc_pages_nodemask) from [<802abde0>] (pagecache_get_page+0x10c/0x34c)
[289804.364079]  r10:80f316db r9:00000006 r8:80e059e4 r7:898314e8 r6:00100cca r5:00000044
[289804.364083]  r4:17240003
[289804.364090] [<802abcd4>] (pagecache_get_page) from [<802add40>] (filemap_fault+0x638/0xb20)
[289804.364097]  r10:80e05008 r9:89579ec8 r8:00000040 r7:834d76c0 r6:89579ec8 r5:834d76c0
[289804.364101]  r4:898313d8
[289804.364110] [<802ad708>] (filemap_fault) from [<80427a9c>] (ext4_filemap_fault+0x38/0x4c)
[289804.364117]  r10:89579fb0 r9:89579ec8 r8:00000040 r7:00000000 r6:75774000 r5:89579ec8
[289804.364120]  r4:898313c0
[289804.364130] [<80427a64>] (ext4_filemap_fault) from [<802e4f04>] (__do_fault+0x48/0x14c)
[289804.364134]  r5:894200c0 r4:89579ec8
[289804.364142] [<802e4ebc>] (__do_fault) from [<802e9fc0>] (handle_mm_fault+0xae8/0xe30)
[289804.364148]  r7:00000000 r6:75774000 r5:80e05008 r4:00000254
[... 73 lines cut ...]
[289804.364870] [    941]     0   941     2561       83    18432       41             0 sudo
[289804.364879] [    945]     0   945     2156      278    16384        4             0 bash
[289804.364889] [  27015]     0 27015     1966      156    14336        9             0 man
[289804.364897] [  27025]     0 27025     1689       35    14336        5             0 pager
[289804.364909] [  23390]     0 23390      485       25    10240        0             0 apt.systemd.dai
[289804.364918] [  23400]     0 23400      485       25     8192        0             0 apt.systemd.dai
[289804.364927] [  23411]     0 23411    13027     2026    61440        0             0 apt-get
[289804.364936] [  23429]     0 23429     2367       77    20480        5             0 cron
[289804.364944] [  23434]     0 23434      485       17     8192        0             0 sh
[289804.364953] [  23436]     0 23436      465       20    12288        0             0 run-parts
[289804.364962] [  23439]     0 23439      485       18    12288        0             0 dpkg
[289804.364971] [  23444]     0 23444      485       24    10240        0             0 savelog
[289804.364982] [  23497]     0 23497      747       26    10240        0             0 mv
[289804.364990] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=octodash,pid=509,uid=1000
[289804.365120] Out of memory: Killed process 509 (octodash) total-vm:817456kB, anon-rss:249080kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:632kB oom_score_adj:0
[289804.469221] oom_reaper: reaped process 509 (octodash), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

```
**General Information:**
 - Hardware: Raspberry Pi 3 A+ and DFR0550 (5'' 800x480 TFT Raspberry Pi DSI Touchscreen)
 - OS Info: OctoPi 0.17.0
 - OctoDash Version v2.2.0
 - OctoPrint Version 1.6.1

**Additional context**

UnchartedBull, 2021-07-08:
Oh damn. I'm going to investigate this as soon as possible and release a fixed version. I'll let you know once I was able to replicate this.

UnchartedBull, 2021-07-08:
Ok I just had a look at this and did a bit of testing. I wasn't able to reproduce this on the dev system. The JS heap size always stayed in the 5-8MB area and also the memory profile doesn't show any obvious signs of a memory leak:

[image]

Could yo maybe elaborate on what you did to cause OctoDash to fail? Was it just idleing? According to your logs OctoDash is filling up around half of the memory in about 45 minutes, so it would crash after around 70-80 minutes?

mithkr, 2021-07-09:
The problem is on my system seen immediately after reboot. System is fresh rebooted and octoprint and octodash is started. Octoprint is connected to 3d printer and only temperature sensor values are exchange between printer, octoprint and octodash. Nothing else is started. I made new top measurement after reboot and there are two processes (pid: 510, 576) of octodash are continuous increasing in the RES memory. 
I also add at the end the list of the octodash processes with their args and attached strace files of this two processes (strace -o octodash-510 -tt -ff -p 510; strace -o octodash-576 -tt -ff -p 576)
[octodash-strace.zip](https://github.com/UnchartedBull/OctoDash/files/6793125/octodash-strace.zip)
```
top - 19:06:10 up 3 min,  2 users,  load average: 1,92, 1,47, 0,63
Tasks: 131 total,   1 running, 130 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1,6 us,  1,2 sy,  0,0 ni, 97,2 id,  0,1 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     38,3 free,    197,6 used,    224,7 buff/cache
MiB Swap:    100,0 total,     50,5 free,     49,5 used.    206,7 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                                     
  431 pi        20   0  410264  76412   8008 S   8,3  16,2   1:18.95 octoprint                                                                                                                                                                                   
  510 pi        20   0  580912  43900   7692 S   0,0   9,3   0:05.87 octodash                                                                                                                                                                                    
  576 pi        20   0  309048  42876  18456 S   0,7   9,1   0:11.68 octodash                                                                                                                                                                                    
  592 pi        20   0  215040  18748   8544 S   0,0   4,0   0:02.19 octodash                                                                                                                                                                                    
  557 pi        20   0  209476  10924   5056 S   0,3   2,3   0:00.59 octodash 

  top - 19:08:34 up 6 min,  2 users,  load average: 0,27, 0,93, 0,54
Tasks: 125 total,   1 running, 124 sleeping,   0 stopped,   0 zombie
%Cpu(s):  2,2 us,  0,7 sy,  0,0 ni, 97,0 id,  0,1 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     33,2 free,    198,8 used,    228,7 buff/cache
MiB Swap:    100,0 total,     50,5 free,     49,5 used.    205,4 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                                     
  431 pi        20   0  410264  76412   8008 S   8,3  16,2   1:31.23 octoprint                                                                                                                                                                                   
  510 pi        20   0  580912  52180  15808 S   0,3  11,1   0:06.36 octodash                                                                                                                                                                                    
  576 pi        20   0  309048  43180  18840 S   0,3   9,2   0:12.53 octodash                                                                                                                                                                                    
  592 pi        20   0  215040  18748   8544 S   0,0   4,0   0:02.19 octodash                                                                                                                                                                                    
  557 pi        20   0  209476  10924   5056 S   0,3   2,3   0:00.78 octodash

  top - 19:11:36 up 9 min,  2 users,  load average: 0,08, 0,53, 0,45
Tasks: 124 total,   1 running, 123 sleeping,   0 stopped,   0 zombie
%Cpu(s):  2,0 us,  0,7 sy,  0,0 ni, 97,3 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     31,3 free,    199,2 used,    230,2 buff/cache
MiB Swap:    100,0 total,     50,5 free,     49,5 used.    205,0 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                                                                     
  431 pi        20   0  410264  76412   8008 S   8,6  16,2   1:46.65 octoprint                                                                                                                                                                                   
  510 pi        20   0  580912  52180  15808 S   0,3  11,1   0:06.84 octodash                                                                                                                                                                                    
[... cut ...] res=CertVerifierService,CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCookies,SpareRendererForSitePerProcess --gpu-preferences=UAAAAAAAAAAgAAAQAAAAAAAAAAAAAAAAAABgAAAAAAAwAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgAAAAAAAAACAAAAAAAAAAAAAAAAAAAAAIAAAAAAAAAAgAAAAAAAAACAAAAAAAAAA= --use-gl=swiftshader-webgl --shared-files
pi         594  0.0  0.6 171084  3228 tty1     S    19:02   0:00 /opt/OctoDash/octodash --type=broker
root      1631  0.0  0.1   7360   504 pts/1    S+   19:24   0:00 grep octodash
```

UnchartedBull, 2021-07-16:
Thanks for the additional info! I'll do some further investigation and probably provide a dev build here to test.

mithkr, 2021-07-22:
I have made a further investigation and looked how octodash behave if printer is disconnected from octoprint.
If printer is disconnected, than octodash show sleep screen and memory leak is not seen. So memory leak maybe somehow related to the printer status update messages (e.g. temp sensors) between octodash and octoprint.
Here the top measurement with disconnected printer:
```
top - 22:43:06 up 1 day, 12:27,  2 users,  load average: 0,04, 0,06, 0,05
Tasks:   8 total,   0 running,   8 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1,4 us,  0,6 sy,  0,0 ni, 97,9 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     44,6 free,    210,3 used,    205,8 buff/cache
MiB Swap:    100,0 total,     66,5 free,     33,5 used.    191,0 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
  580 pi        20   0  339864  76724  36328 S   0,3  16,3   6:59.93 /opt/OctoDash/octodash --type=renderer --touch-events=enabled --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,+
  510 pi        20   0  573872  61488  31660 S   0,2  13,0   4:22.77 /opt/OctoDash/octodash
  591 pi        20   0  217524  28016  14388 S   0,0   5,9   0:02.39 /opt/OctoDash/octodash --type=gpu-process --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCoo+
  558 pi        20   0  209580  17900  12248 S   0,1   3,8   2:15.92 /opt/OctoDash/octodash --type=utility --utility-sub-type=network.mojom.NetworkService --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWith+
  515 pi        20   0  158712   9484   4640 S   0,0   2,0   0:00.13 /opt/OctoDash/octodash --type=zygote
  514 pi        20   0  158712   9392   4576 S   0,0   2,0   0:00.14 /opt/OctoDash/octodash --type=zygote --no-zygote-sandbox
  598 pi        20   0  173568   6992   1756 S   0,0   1,5   0:00.00 /opt/OctoDash/octodash --type=broker
  517 pi        20   0  158712   4984    144 S   0,0   1,1   0:00.00 /opt/OctoDash/octodash --type=zygote

top - 01:43:08 up 2 days, 15:27,  2 users,  load average: 0,00, 0,03, 0,05
Tasks:   8 total,   0 running,   8 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1,4 us,  0,6 sy,  0,0 ni, 98,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     49,7 free,    212,6 used,    198,3 buff/cache
MiB Swap:    100,0 total,     63,5 free,     36,5 used.    185,5 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
  580 pi        20   0  339864  79320  36344 S   0,3  16,8  11:59.53 /opt/OctoDash/octodash --type=renderer --touch-events=enabled --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,+
  510 pi        20   0  573872  59028  31652 S   0,2  12,5   7:33.04 /opt/OctoDash/octodash
  591 pi        20   0  217524  28012  14384 S   0,0   5,9   0:02.67 /opt/OctoDash/octodash --type=gpu-process --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCoo+
  558 pi        20   0  225972  18856  13016 S   0,1   4,0   3:56.48 /opt/OctoDash/octodash --type=utility --utility-sub-type=network.mojom.NetworkService --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWith+
  515 pi        20   0  158712   9480   4636 S   0,0   2,0   0:00.13 /opt/OctoDash/octodash --type=zygote
  514 pi        20   0  158712   9388   4572 S   0,0   2,0   0:00.14 /opt/OctoDash/octodash --type=zygote --no-zygote-sandbox
  598 pi        20   0  173568   6992   1756 S   0,0   1,5   0:00.00 /opt/OctoDash/octodash --type=broker
  517 pi        20   0  158712   4984    144 S   0,0   1,1   0:00.00 /opt/OctoDash/octodash --type=zygote

top - 15:43:09 up 3 days,  5:27,  2 users,  load average: 0,02, 0,08, 0,03
Tasks:   8 total,   0 running,   8 sleeping,   0 stopped,   0 zombie
%Cpu(s):  1,4 us,  0,6 sy,  0,0 ni, 98,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :    460,7 total,     49,5 free,    212,1 used,    199,1 buff/cache
MiB Swap:    100,0 total,     63,5 free,     36,5 used.    186,0 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
  580 pi        20   0  339864  79616  36340 S   0,3  16,9  14:35.27 /opt/OctoDash/octodash --type=renderer --touch-events=enabled --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,+
  510 pi        20   0  573872  58928  31612 S   0,2  12,5   9:11.75 /opt/OctoDash/octodash
  591 pi        20   0  217524  28012  14384 S   0,0   5,9   0:02.82 /opt/OctoDash/octodash --type=gpu-process --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWithoutSameSiteMustBeSecure,SameSiteByDefaultCoo+
  558 pi        20   0  225972  18972  13016 S   0,1   4,0   4:48.81 /opt/OctoDash/octodash --type=utility --utility-sub-type=network.mojom.NetworkService --field-trial-handle=12013433723393411160,11906847453383645420,131072 --disable-features=CookiesWith+
  515 pi        20   0  158712   9480   4636 S   0,0   2,0   0:00.13 /opt/OctoDash/octodash --type=zygote
  514 pi        20   0  158712   9388   4572 S   0,0   2,0   0:00.14 /opt/OctoDash/octodash --type=zygote --no-zygote-sandbox
  598 pi        20   0  173568   6992   1756 S   0,0   1,5   0:00.00 /opt/OctoDash/octodash --type=broker
  517 pi        20   0  158712   4984    144 S   0,0   1,1   0:00.00 /opt/OctoDash/octodash --type=zygote

```

UnchartedBull, 2021-07-25:
Interesting find. I'm going to post a dev build here sometime tomorrow, please try that one as well and report whether the memory leak is gone. Thanks!

I also don't know what you changed with your top command in your latest comment, but having the arguments for each process actually is super useful, so please include them for further tests if possible! :)

UnchartedBull, 2021-07-26:
https://drive.google.com/file/d/1M1_Ksvj49Naxz-Yijxj2pmalApb07Bl0/view?usp=sharing please try this version and report whether this fixes the issue.

Also i noticed, that your raspberry only has 512MB of memory, which model are you running this on?

mithkr, 2021-07-26:
Thanks, I will test it.

I have a raspberry pi 3A+

mithkr, 2021-07-28:
I have run your test version for 40 hour and capture memory usage every 10 minutes with following command:
`top -o RES -p $(pidof -f %d, octodash)0 -b -c -d 600 | tee top.log`
unluckily mem leak is still there. See attached top.log and generated graphs from that log file. top_plot.txt contains the script I used to generate the graphs from the top.log file.
[top.log](https://github.com/UnchartedBull/OctoDash/files/6891730/top.log)
[image]
[image]
[image]
[top_plot.txt](https://github.com/UnchartedBull/OctoDash/files/6891850/top_plot.txt)

UnchartedBull, 2021-08-02:
Could you try this version: https://drive.google.com/file/d/1uTu9q4XjojOWRfrwU4-win52v4AuwVY0/view?usp=sharing and report whether that fixed the issue?

mithkr, 2021-08-04:
I have tested. I looks like that after some time memory increase stops now, see charts. I will further test after my vacation in 3 weeks.
[image]
[image]
[image]
[top.log](https://github.com/UnchartedBull/OctoDash/files/6928890/top.log)

UnchartedBull, 2021-08-04:
Ok that looks like we're finally making progress! Thanks for the all graphs. It's really weird that it's allocating memory for a while and then just stops. Might need to do some further digging.

Enjoy your vacation!

meganssmith, 2021-09-18:
I have also been sealing with this. Any idea when the fix will go main stream?

UnchartedBull, 2021-09-22:
Once @mithkr has confirmed that this has been fixed for him :) But for now he should enjoy his vacation :) Will definitely be included in the next release though.

mithkr, 2021-09-27:
I have now started a long time recording of top, to see memory consumption for 1 week. I will then upload the result.

mithkr, 2021-10-05:
Here are the test results. It looks good. Measurement period is about 9 days. No OOM happened. I only see a very little increase in the virtual memory chart (light blue). Due to the jumps in the RES memory chart, due to enabled swap, it is not possible to proof this trend there. I will repeat measurement with disabled swap. But it can be concluded, that the fixes you already provided helped a lot.
[top.log](https://github.com/UnchartedBull/OctoDash/files/7286123/top.log)
[image]
[image]
[image]

UnchartedBull, 2021-10-09:
Ok great thanks for the update! If anything weird shows up again feel free to re-open this issue :)

closed 2021-10-09 by UnchartedBull