UnchartedBull/OctoDash #2036: PSU control wont wake printer after V2.2.0 update

bach37strad, 2021-07-10:
raspberry pi 3b+ / Ender 3 V2 (4.2.7)
Optopi 0.17.0 / Octoprint 1.6.1 / Octodash V2.2.0

After updating to V2.2.0 from V2.0.0 the powertoggle function in custom actions will no longer wake the printer, only turn it off. Sleep mode screen has not worked since installing klipper however I was always able to navigate to the control screen to turn the printer on. PSU control still works normally through octoprint's desktop UI.

bach37strad, 2021-07-10:
[octoprint.log](https://github.com/UnchartedBull/OctoDash/files/6795867/octoprint.log)
[klippy.log](https://github.com/UnchartedBull/OctoDash/files/6795868/klippy.log)
[printer.cfg.txt](https://github.com/UnchartedBull/OctoDash/files/6795869/printer.cfg.txt)
[octoprint.log.txt](https://github.com/UnchartedBull/OctoDash/files/6795872/octoprint.log.txt)

bach37strad, 2021-07-10:
Please ignore the "unable to open port" connection issue stuff. That was a klipper issue that's since been resolved.

UnchartedBull, 2021-07-16:
Did OctoPrint show any errors? Did you enable multiple PSU Control Plugins with v2.2.0?

bach37strad, 2021-07-16:
No errors in octoprint. It's like the commands dont even register until the printer is powered on. As far as PSU control plugins, I only have the one instance installed and have not changed anything since the initial setup.
[image]

bach37strad, 2021-07-16:
I have noticed sometimes using the custom home action ["G28; G91; G0 X-40 F3000; G90; G0 Z0 F200"] will wake the printer, but not reliably, and it does not home after waking until i run the command again.

UnchartedBull, 2021-07-25:
`!POWERTOGGLE` is an octodash only command, so the error here is expected.

From what I can see in the logs it seems like OctoPrint is having a hard time connecting to your printer `tmp/printer`. Please double check your connection settings and ideally post a screenshot of that part of the UI (once while connected and once while sleeping)

closed 2021-08-24 by stale[bot]

jrydval, 2021-08-28:
I am facing the same problem - 
The "command": "[!POWERTOGGLE]" is only able to switch my printer off (via PSU Control plugin) but not on when pressed on the "sleep screen".

- OctoDash 2.2.0
- OctoPrint 1.7.0rc2

I found these errors in octoprint.log file:

```
2021-08-28 12:50:00,718 - tornado.application - ERROR - Exception in callback functools.partial(<function RawWebSocketTransport.send_pack.<locals>.<lambda> at 0xa60b1ae0>)
Traceback (most recent call last):
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/tornado/ioloop.py", line 741, in _run_callback
    ret = callback()
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 90, in <lambda>
    self.server.io_loop.add_callback(lambda: self.send_pack(message, binary))
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 85, in send_pack
    self.write_message(message, binary).add_done_callback(self.send_complete)
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/tornado/websocket.py", line 339, in write_message
    message = tornado.escape.json_encode(message)
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/octoprint/server/util/tornado.py", line 48, in fixed_json_encode
    return json.dumps(value, default=JsonEncoding.encode, allow_nan=False).replace(
  File "/usr/lib/python3.7/json/__init__.py", line 238, in dumps
    **kw).encode(obj)
  File "/usr/lib/python3.7/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.7/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/home/pi/Apps/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/json/__init__.py", line 49, in encode
    raise TypeError
TypeError
```

These ERRORs are new ones after octodash installation but to be honest I am not 100% sure that they are related to this problem. But they appear around my attempts to switch the printer on.

Best Regards,
Jan

jneilliii, 2021-08-28:
This could be an issue introduced by my companion plugin @jrydval ? I had a couple of issues with some of the plugin settings early on. Are you using that plugin?

jrydval, 2021-08-28:
Hi there! Yes, it's installed but actually I do not need it anymore. I edit the config file directly. I will uninstall it and we will see...

Jan

jneilliii, 2021-08-28:
would need to know which field got messed up if you don't mind. not sure what could have triggered the type error.