UnchartedBull/OctoDash #2061: Flickering Display

nicedevil007, 2021-07-18:
**Describe the bug**
I have a Biqu BX and the display starts flickering after I switch to Octodash. On normal Marlin Mode or Touch UI everything seems to be fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Octodash as described in the Wiki
2. Switch to Octodash
3. Starting flickering

**Expected behavior**
No flickering at all

**General Information:**
 - Raspberry Pi 4
 - OctoPi (latest official built)
 - OctoDash Version (latest official built)
 - OctoPrint Version (latest official built)

BiquOwner, 2021-07-25:
I just setup my Biqu Bx just recently. 
I setup OctoDash because of Teaching Techs instructions on Youtube. https://www.youtube.com/watch?v=bfhhUVJkP_s

You need to follow these instructions to setup the screen properly. I had this exact problem.
https://github.com/bigtreetech/OctoBTT
----------------------
*    If you are a BX machine model user, you will need to add the following fragment lines to the RaspberryPi boot/config.txt file to display them correctly on screen:

        max_usb_current=1
        hdmi_group=2
        hdmi_mode=87
        hdmi_cvt 1024 600 60
        hdmi_drive=1
-------------------------

Good Luck :)

nicedevil007, 2021-07-25:
That fixed everything! Thank you, no hardware issue!

closed 2021-07-25 by nicedevil007

UnchartedBull, 2021-07-25:
Thanks for the help @BiquOwner! And great that everything is working now as it should :)