UnchartedBull/OctoDash #2069: Initializing... forever

sergioiker, 2021-07-21:
**What doesn't work?**
Updated Octodash to 2.2.1 And after rebooting, is initializing without success.

**What did you already try?**
I check and uncheck the CORS  option. still not working

**General Information:**
 - Hardware Raspberry pi 4b 
 - OS Info raspi octopi
 - OctoDash Version 2.2.1
 - OctoPrint Version 1.6.1

**Additional context**
help!

UnchartedBull, 2021-07-22:
Did you reboot after checking the CORS checkbox?

Jack77777777, 2021-07-22:
Is Octodash 2.2.1 released ?

UnchartedBull, 2021-07-25:
Not yet, do you think that this would have an impact on this issue?

mustang5269, 2021-07-30:
I had the same issue and had to delete my config and redo it.

gale9010, 2021-08-19:
Same problem. 

I try to connect, unconnect, restart, change api key... 

Any solution?

closed 2021-09-07 by stale[bot]

jiveman, 2023-12-27:
After trying to manually edit the ~/.config/octodash/config.json file with a few different local hostnames, I gave up and uninstalled wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/remove.sh | bash then reinstalled (no-reboot required) bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh) which did make me setup again, but it did work after. CORS was enabled on my instance, and did try disabling/re-enabling.