UnchartedBull/OctoDash #2086: Octodash / Octoprint and Klipper

ThaNeub, 2021-07-26:
Hi,

I recently portet my Ender 3V2 to Klipper, so my LCD became useless.
I bought a BTT PITFT50 to have a Display on my Printer via the PI 4B. With Fluidd and Klipperscreen it works pretty well, unfortunately it doesnt work with Octoprint, Klipper and Octodash.

Here is the problem, after I installed the Octodash Package as shown in this github and restartet the PI, my Printer won´t connect to Octoprint with Klipper anymore saying that it could not establish a connection. (Multiple connections on Port)

Is there a known Fix for this? I want to use Octoprint, cause it seems to be more comfortable than Fluidd

ThaNeub

UnchartedBull, 2021-07-28:
Can you connect via the normal Octoprint UI or is that not working as well?

UnchartedBull, 2021-08-12:
Then you first need to fix the OctoPrint connection. I can't provide any help with that though. Probably best to browse around the internet. 

Once you've fixed your issue with the OctoPrint Connection and OctoDash still doesn't work please reopen this issue.