UnchartedBull/OctoDash #2090: I can’t change the theme

eacxx, 2021-07-26:
**What doesn't work?**
I’m replacing the custom-styles.css file, but the layout of Octodash remains the default one.

**What did you already try?**
I used any theme file that I found. I’ve even deleted the one in the config folder. Useless. The Octodash is the same…

**General Information:**
 - Hardware: Raspberry PI Model B Rev 1.4, 2GB RAM, 5” display, 800x600
 - OS Info: OctoPi 0.18.0
 - OctoDash Version: 2.2.0
 - OctoPrint Version: 1.6.1

jneilliii, 2021-07-26:
Did you restart after changing the file? That is required.

eacxx, 2021-07-27:
Yes man! Each time.
And not only Octodash, many times I've restarted the Raspberry PI entirely.

Jack77777777, 2021-07-27:
Copied to right folder with user pi ?

eacxx, 2021-07-27:
/home/pi/.config/octodash

eacxx, 2021-07-27:
Firstly I’ve tried from a Mac terminal with wget, being logged in with the user pi
Then I’ve tried from a Windows machine with WinSCP, using the same account - pi
The owner is the user pi, both before and after copy.

eacxx, 2021-07-28:
Is there any advice for me?

UnchartedBull, 2021-07-28:
Have you checked the contents of the file after changing the content (`cat custom-styles.css`)? Please also make sure that the name is 100% the same. OctoDash will only create the file if it either can't open it (it isn't present).

eacxx, 2021-07-28:
I said Octodash created the empty file only when I have deleted it. And I've did this only because I wanted to see if I'm looking in the right directory or not.

UnchartedBull, 2021-07-28:
No problem, GitHub can be a bit tricky at times. Glad you figured it out :)

closed 2021-07-28 by UnchartedBull