UnchartedBull/OctoDash #2114: Completion Time Option vs Hours Left

zero1niner, 2021-08-03:
Would love the option to set either the 'time left' and 'time elapsed' OR the actual system time which the job will finish.

Perfect world would be to be able to choose 2 of the options to fit in that location. 

Personally, Im always more interested in the time it will complete (without doing the math...im lazy that way). Helps me plan on the next step. 

Seems like some easy match to add the time left to the pi system time to produce the value. 

Amazing product. SUPER clean look is what I love the most. You have a great eye for UI/UX design.

thegarbz, 2021-08-12:
The est. finishing time shouldn't change during the print unless you play with something like feed rate. This time is actually displayed if you load a file without clicking print yet, (though I believe this is an estimate and not read out of the g-code). I'll see if I can do something with it in a theme but I suspect not without something added to the underlying code.

jneilliii, 2021-08-12:
My guess is all the time estimates are from file analysis, which if you auto start a print on upload from prusaslicer that analysis doesn't happen.

fmdev1, 2021-08-12:
So the actual completion time would be a simple math formula of current system time + hours remaining to print. Even if its off a few minutes for whatever reason, it would still be super handy. I use an app called Octopod on my iphone that shows the completion time, and love having that at a quick glance.

D-Bongard, 2023-07-12:
On the same sort of topic, why are you able to speed up the print but the time remaining does not recalculate?
I can literally double the print speed and the hours remain ticks down as though I hadn't...but the print finishes when the total print time is only half used.  Even if it doesn't re-calc everything in the code it could at least maybe apply the higher print rate percentage to the total print time and display that to try to get closer to the mark???

jneilliii, 2023-07-12:
That's because feed rate increases aren't taking into account in the calculations I suspect.

D-Bongard, 2023-07-12:
Well for the same reason as this thread was created....to know the complation time....I would suspect that they very much should.
My printer runs better (smoother surfaces) slowly, but can run quick when it needs to, and so I want to adjust my feedrate to be able to take my overnight prints to work in the morning.  But as it is I just have to do rough calcs using the percentage and trying to account for the time already taken.  But doing this accurately should be child's play for Octodash!?!?!