UnchartedBull/OctoDash #2139: PSU Control and octodash..

jbuszkie, 2021-08-06:
**What doesn't work?**
Octodash doesn't seem to turn on the printer when I touch/wakeup the screen.

**What did you already try?**

Fresh install

**General Information:**
PI4.  Should be latest everything.

**Additional context**

I had to do a fresh install due to a failed SD card! *ugh*
I'm sure I'm probably missing something in PSU config in the web UI..
Power functions work fine from the web UI.

This worked a while ago..  But I don't remember how I got it working.
I'm sure it's something stupid I'm missing!

jneilliii, 2021-08-06:
Did you enable the option and tell the plugin that you have that plugin installed?

jbuszkie, 2021-08-06:
I *thought* I did.  I swore I enabled it when I installed...  It turns out it wasn't.
Problem #1 fixed.
But it still didn't work.
I also needed to set the "turn on printer when exiting sleep"
Problem #2 fixed.

I seem to be working fine now.
I knew I would probably feel stupid when I figured out the problem!!

Thanks for the nudge in the right direction!

Now how to figure out how to turn my light on and off from octo dash! :-)

Jim

closed 2021-08-06 by jbuszkie

jneilliii, 2021-08-06:
Use my companion plugin and you can add the custom actions easily to give you a button to do so.

jbuszkie, 2021-08-06:
> Use my companion plugin and you can add the custom actions easily to give you a button to do so.

OK..  Which companion plugin is that?  It looks like the !OUTPUT isn't supported yet?

Jim

jneilliii, 2021-08-06:
The OctoDash Companion plugin for OctoPrint. I wasn't aware of `!OUTPUT`, is that new?

jneilliii, 2021-08-06:
you can always just put the text in manually...

jneilliii, 2021-08-06:
I've just added those to a release candidate. Change the plugin's release channel in OctoPrint's Software Update settings to release Candidate and upgrade when prompted.

UnchartedBull, 2021-08-07:
!OUPUT (and !OUPUT_PWM) haven't been released yet. The code is merged and they'll work if you build manually from master, but v2.2.0 doesn't support those commands. Will be released with v2.3.0

jneilliii, 2021-08-07:
Thanks @UnchartedBull, I'll hold full release until 2.3.0 comes out. Let me know if you want any other websocket data added to the plugin that's not currently available if you need to tweak anything.