UnchartedBull/OctoDash #2165: Stuck on Hey There screen

rgillasp, 2021-08-17:
Gets to "Hey There" screen, tap on "Next", Octoprint instance is shown, but cannot select it. Acts like entire touchscreen is dead except for the back button.

Tried fresh install of Octoprint, Installed, Removed, and Re-Installed Octodash

**General Information:**
 - Raspberry Pi 3B+, OOSYOO 5" DSL Touchscreen 
 - Octopi 0.18.0
 - OctoDash Latest as of 8-17-2021
 - OctoPrint Version 1.5.2

UnchartedBull, 2021-08-18:
Are you sure that your touchscreen is working as expected? Probably worth a try to attach a mouse to go through the wizard or try setting up the config manually https://github.com/UnchartedBull/OctoDash/wiki/Setup#creating-the-config-manually.

rgillasp, 2021-08-18:
That was it! Thank you so much!

closed 2021-08-18 by rgillasp