UnchartedBull/OctoDash #2206: Install issues

clue3030, 2021-08-28:
why do I get this back when I try to install octodash?
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'

Will-wastelander, 2021-08-28:
This is related to APT, not OctoDash. You would get this regard of what you are trying to install via APT.

Will-wastelander, 2021-08-28:
```
pi@Biqu-BX-octopi:~ $ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
Installing OctoDash v2.2.0, armv7l
Installing Dependencies ...
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
pi@Biqu-BX-octopi:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] y
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Fetched 13.2 MB in 19s (690 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
140 packages can be upgraded. Run 'apt list --upgradable' to see them.
pi@Biqu-BX-octopi:~ $ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
Installing OctoDash v2.2.0, armv7l
Installing Dependencies ...
140 packages can be upgraded. Run 'apt list --upgradable' to see them.
libuuid1 is already the newest version (2.33.1-0.1).
libxss1 is already the newest version (1:1.2.3-1).
libxss1 set to manually installed.
libxtst6 is already the newest version (2:1.2.3-1).
libxtst6 set to manually installed.
The following additional packages will be installed:
  9menu adwaita-icon-theme at-spi2-core dbus-user-session dconf-gsettings-backend dconf-service fonts-dejavu fonts-dejavu-extra fonts-vlgothic glib-networking glib-networking-common glib-networking-services gsettings-desktop-schemas
  gtk-update-icon-cache libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libauthen-sasl-perl libcairo-gobject2 libcolord2 libcups2 libdata-dump-perl libdbusmenu-glib4 libdbusmenu-gtk3-4 libdconf1 libegl-mesa0 libegl1 libegl1-mesa
  libencode-locale-perl libepoxy0 libevdev2 libfile-basedir-perl libfile-desktopentry-perl libfile-listing-perl libfile-mimeinfo-perl libfont-afm-perl libfontenc1 libgbm1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 libgtk-3-bin
  libgtk-3-common libgudev-1.0-0 libhtml-form-perl libhtml-format-perl libhtml-parser-perl libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl libhttp-daemon-perl libhttp-date-perl libhttp-message-perl libhttp-negotiate-perl
  libindicator3-7 libinput-bin libinput10 libio-html-perl libio-socket-ssl-perl libio-stringy-perl libipc-system-simple-perl libjson-glib-1.0-0 libjson-glib-1.0-common liblwp-mediatypes-perl liblwp-protocol-https-perl libmailtools-perl
  libmtdev1 libnet-dbus-perl libnet-http-perl libnet-smtp-ssl-perl libnet-ssleay-perl libnspr4 libpciaccess0 libproxy1v5 libpthread-stubs0-dev librest-0.7-0 librsvg2-common libsecret-common libsoup-gnome2.4-1 libsoup2.4-1
  libstartup-notification0 libtie-ixhash-perl libtimedate-perl libtry-tiny-perl libunwind8 liburi-perl libwacom-bin libwacom-common libwacom2 libwayland-server0 libwww-perl libwww-robotrules-perl libx11-6 libx11-protocol-perl
  libxau-dev libxaw7 libxcb-util0 libxcb1-dev libxcomposite1 libxdmcp-dev libxfont2 libxft2 libxkbfile1 libxml-parser-perl libxml-twig-perl libxml-xpathengine-perl libxmu6 libxt6 libxxf86dga1 menu notification-daemon
  perl-openssl-defaults rxvt rxvt-unicode x11-utils x11-xkb-utils x11proto-core-dev x11proto-dev xfonts-base xfonts-encodings xfonts-utils xorg-sgml-doctools xserver-common xserver-xorg-core xserver-xorg-input-all
  xserver-xorg-input-libinput xserver-xorg-input-wacom xserver-xorg-legacy xserver-xorg-video-all xserver-xorg-video-amdgpu xserver-xorg-video-ati xserver-xorg-video-fbdev xserver-xorg-video-nouveau xserver-xorg-video-radeon
  xserver-xorg-video-vesa xtrans-dev
Suggested packages:
  libdigest-hmac-perl libgssapi-perl colord cups-common gvfs libcrypt-ssleay-perl libauthen-ntlm-perl libx11-doc libxcb-doc libunicode-map8-perl libunicode-string-perl xml-twig-tools menu-l10n gksu | kde-runtime | ktsuss xbindkeys
  xclip mesa-utils nickle cairo-5c xorg-docs-core xfonts-100dpi | xfonts-75dpi xfonts-scalable xinput firmware-amd-graphics xserver-xorg-video-r128 xserver-xorg-video-mach64
The following NEW packages will be installed:
  9menu adwaita-icon-theme at-spi2-core bc dbus-user-session dconf-gsettings-backend dconf-service desktop-file-utils fonts-dejavu fonts-dejavu-extra fonts-vlgothic glib-networking glib-networking-common glib-networking-services
  gsettings-desktop-schemas gtk-update-icon-cache libappindicator3-1 libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libauthen-sasl-perl libcairo-gobject2 libcolord2 libcups2 libdata-dump-perl libdbusmenu-glib4
  libdbusmenu-gtk3-4 libdconf1 libegl-mesa0 libegl1 libegl1-mesa libencode-locale-perl libepoxy0 libevdev2 libfile-basedir-perl libfile-desktopentry-perl libfile-listing-perl libfile-mimeinfo-perl libfont-afm-perl libfontenc1 libgbm1
  libgtk-3-0 libgtk-3-bin libgtk-3-common libgudev-1.0-0 libhtml-form-perl libhtml-format-perl libhtml-parser-perl libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl libhttp-daemon-perl libhttp-date-perl libhttp-message-perl
[... cut ...]
  xfonts-encodings xfonts-utils xinit xorg-sgml-doctools xserver-common xserver-xorg xserver-xorg-core xserver-xorg-input-all xserver-xorg-input-libinput xserver-xorg-input-wacom xserver-xorg-legacy xserver-xorg-video-all
  xserver-xorg-video-amdgpu xserver-xorg-video-ati xserver-xorg-video-fbdev xserver-xorg-video-nouveau xserver-xorg-video-radeon xserver-xorg-video-vesa xtrans-dev
The following packages will be upgraded:
  libavahi-compat-libdnssd1 libgl1-mesa-dri libglapi-mesa libglx-mesa0 libx11-6
5 upgraded, 147 newly installed, 0 to remove and 135 not upgraded.
Need to get 55.7 MB of archives.
After this operation, 138 MB of additional disk space will be used.

```

clue3030, 2021-08-28:
What’s APT? I installed octoprint on my rpi4. I’m on a mac. I go into terminal and then ssh into pi then I and put this in: bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh) and then that comes up and then goes back to the prompt. What do I need to do to install this through a Mac?

Will-wastelander, 2021-08-29:
I gave you the commands to run if you look at what I pasted. APT is the linux software package tool. It is used to install, update, remove, etc all programs for the OS. Check the links for further info.

https://letmegooglethat.com/?q=apt+raspberry+pi
https://letmegooglethat.com/?q=E%3A+Repository+%27http%3A%2F%2Fraspbian.raspberrypi.org%2Fraspbian+buster+InRelease%27+changed+its+%27Suite%27+value+from+%27stable%27+to+%27oldstable%27

clue3030, 2021-08-29:
I’m new to octodash and octoprint. I only know the basics on how to install octoprint and run it off my MacBook Pro. Other than that I’m lost to other things with octoprint.

Will-wastelander, 2021-08-29:
This has nothing to do w/ OctoPrint either. This is a OS related problem. Look at the output I posted, and follow the commands after the error'd command.

Should be sudo apt update, and then answer the question(s). then you should be able to install OctoDash. If you are going to be using a Linux OS, you should read up on how to manage it, so you are not leaving your network open to security risks.

clue3030, 2021-08-29:
Yes that worked. It’s now installed. Thank you so very much.

Will-wastelander, 2021-08-29:
I'm doing the same thing right now.

jneilliii, 2021-08-29:
So I just submitted a PR that may resolve this in the install script. It's related to an upstream debian release name change or something. We figured this out with CustoPiZer creation and the fix their resolved the same error for us.

jneilliii, 2021-08-29:
in the meantime, this command should in theory work...
`bash <(wget -qO- https://raw.githubusercontent.com/jneilliii/OctoDash/patch-1/scripts/install.sh)`

shueja-personal, 2021-08-29:
The mentioned PR change of adding --allow-releaseinfo-change worked for me when done by manually editing the script between download and execution.

clue3030, 2021-08-29:
I was able to get it to work by adding a command in terminal : sudo apt update

UnchartedBull, 2021-09-16:
The install script will now show an error message and a quick fix for this issue. Closing this here now.

closed 2021-09-16 by UnchartedBull