UnchartedBull/OctoDash #2207: install issues

clue3030, 2021-08-28:
This is what I get when I try to install it and then it goes back to prompt
"E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'"

shueja-personal, 2021-08-29:
This is a known bug with the way Raspbian handles installing things and a status change on one of the places it looks. It's not strictly an OctoDash thing. I'm testing with the workaround we've used at CustoPiZer.

jneilliii, 2021-08-29:
So I just submitted a PR that may resolve this in the install script. It's related to an upstream debian release name change or something. We figured this out with CustoPiZer creation and the fix their resolved the same error for us.

in the meantime, this command should in theory work...
`bash <(wget -qO- https://raw.githubusercontent.com/jneilliii/OctoDash/patch-1/scripts/install.sh)`

jneilliii, 2021-09-04:
duplicate of #2206

closed 2021-09-08 by UnchartedBull