UnchartedBull/OctoDash #2217: error of InRelease' changed its 'Suite' value from 'stable' to 'oldstable

Tenkiller77, 2021-08-31:
I installed Octo-dash on a pi4 a few weeks ago with no issues. Now I tried setting up another pi4 and now I get this error: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable . How do I fix this? I get this same error when I try to setup octoscreen.

jneilliii, 2021-09-01:
You should be able to run `sudo apt update` before running the install script and it should work based on what others have said. Otherwise you can try using my patched install script. 

```
bash <(wget -qO- https://raw.githubusercontent.com/jneilliii/OctoDash/patch-1/scripts/install.sh)
```

jneilliii, 2021-09-04:
duplicate of #2206

Tenkiller77, 2021-09-07:
I did that, rebooted and tried to install octodash I get the same error
enfact I get the same error when I sudo apt-get update so maybe this is a
octoprint problem.

closed 2021-09-08 by UnchartedBull