UnchartedBull/OctoDash #2233: E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable'

zrox2k, 2021-09-04:
**What doesn't work?**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**What did you already try?**
A clear and concise description of what you tried to make OctoDash work.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context or screenshots about the feature request here.

jneilliii, 2021-09-04:
Run `sudo apt update` and then try running the install script again.

jneilliii, 2021-09-04:
duplicate of #2217

closed 2021-09-08 by UnchartedBull