UnchartedBull/OctoDash #2237: Stuck on connecting…

clue3030, 2021-09-07:
When I turn on my RPI4 and tap the screen to bring octodash out of sleeping, it says on connecting and it doesn't move on the the reg GUI? Is there a workaround or something? [image]

UnchartedBull, 2021-09-08:
Is OctoPrint connected to your printer?

jrydval, 2021-09-14:
I have the same problem - when I turn-on the printer using action icon from the "sleeping screen" then it connects every time. When I turn-on the printer from OctoPrint web UI and then tap the screen to wake-up the OctoDash then OctoDash doesn't connect to the OctoPrint. 

I cannot find any octodash log to look at it. Is there any?

Best Regards,
Jan

jneilliii, 2021-09-14:
Which power plugin are you using @jrydval? Your symptoms seem a little different than others that reported they couldn't get it either way and shortening the auto connect delay resolved it for some users.

jrydval, 2021-09-15:
I use PSUControl plugin with my own on/off/status python scripts controlling TP-Link Tapo P100 plug.

SaKiEQ, 2021-09-21:
I have to say, since the last update (a while ago) i am experiencing the exact same problem, while nothing else has changed on the Raspi or printer. 

I have narrowed it so far down to two things - #1 a problem with Octoprint auto-reconnecting USB to the printer once powered-up, #2 OctoDash not detecting the status change on Octoprint - when the printer is manually reconnected via web UI of Octoprint. 

I would say 50% of the time, the problem is related to OctoPrint no reestablishing the connection, but then still there are cases where everything works fine from an OctoPrint perspective, but only manually killing and restarting OctoDash resolves the issue. 

Further since this connection/wake-from-sleep began, I get every once in a while a problem, where OctoDash stops reporting progress, while still fully responsive on the UI, but the print can neither be canceled nor is the progress updated. 
To further investigate this, i just changed the socket URL in OctoDash from 127.0.0.1 to the machine name 'octopi.local'. Not sure if next would be 'localhost' instead. I can't say for sure, that this is not related to a networking issue on my end.

SaKiEQ, 2021-09-21:
I just solved my issue ! 

The problem presented it first in the serial.log, where Octoprint reports that no port was found to 'auto reconnect' on startup. 
Now I have tried every delay and other timing setting in Octoprint, to make the host wait a few seconds before attempting to connect. The first issue is, that all those settings related to server start/boot not printer start! 

The issue is in the PSU plugin, that requires a 'Post on delay' of about 2 seconds, to wait when detecting active ports. With this delay set, the PSU>On command makes the printer reconnect to OctoPrint reliably, while also reconnecting OctoDash properly 
[image]
!

UnchartedBull, 2021-09-22:
Yes regarding your second issue there still seems to be something wonky going on with the detection. I think @jneilliii had a similar issue when starting prints from PrusaSlicer. This is something that should be fixed in the next release.

Regarding the issue with the freeze during printing: There was an issue open for that, and it has been solved (at least no one reported freezes mid print with that build). Please head over to @jneilliii fork. He has some builds available with the latest fixes. This one should already include that fix: https://github.com/jneilliii/OctoDash/actions/runs/1172339904

SaKiEQ, 2021-09-22:
thanks will sure do !

For me its not a freeze, just the connection seems frozen, Octodash reacts like normal, just does not interact with octoprint. 
If that is what you are referring to, then we are talking about the same :).

UnchartedBull, 2021-09-22:
Yes I did a type in one of the messages that are being send from OctoPrint. Basically if OctoPrint would request a re-authentication OctoDash would just drop that message (which will cause all data to be frozen, but the UI still fully working). So this hopefully will fix this for you as well :)