UnchartedBull/OctoDash #2238: Qidi X-MAX printhead somtimes stops

GN1G, 2021-09-08:
Hello guys!

I have a problem, since I installed OctoDash together wit a BTT pi tft 50 on my Raspberry Pi, the printhead from my Qidi X-MAX somtimes stops for a few seconds and then resumes the print. It happens on different layers and points, were using Octoprint since 1,5 years and had no problem like this.

Does anyone ever had this issue?

Printer: Qidi X-MAX
Raspberry: Pi 4b / 4GB Ram
Display: BTT pi tft 50

thanks for your help!

UnchartedBull, 2021-09-08:
This shouldn't happen, especially not on a Pi4 with 4 gigs of ram. Can you try and remove OctoDash (with the uninstall script) and check whether it's still happening?

Please also provide information about the installed versions (OctoPrint & OctoDash). Thanks!

GN1G, 2021-09-09:
> This shouldn't happen, especially not on a Pi4 with 4 gigs of ram. Can you try and remove OctoDash (with the uninstall script) and check whether it's still happening?
> 
> Please also provide information about the installed versions (OctoPrint & OctoDash). Thanks!

Hey man!

Yeah I think so too.

I removed it this morning with the uninstall script from your website and reinstalled it completely new,
startet a print 5 min ago and it's still happening.

my Octorpint version: 1.6.1
my Octodash version: 2.2.0

GN1G, 2021-09-09:
> This shouldn't happen, especially not on a Pi4 with 4 gigs of ram. Can you try and remove OctoDash (with the uninstall script) and check whether it's still happening?
> 
> Please also provide information about the installed versions (OctoPrint & OctoDash). Thanks!

heres the output from my terminal:

Send: N3828 G1 X213.909 Y156.069 E40.82625*97
Recv: ok N:3828
Send: N3829 G1 X207.302 Y149.461 E41.11905*96
Recv: ok N:3829
Send: N3830 G1 X207.302 Y150.037 E41.13711*110
Recv: wait
Send: N3831 G1 X213.909 Y156.645 E41.42991*108
Recv: ok N:3830
Send: N3832 G1 X213.909 Y157.221 E41.44797*102
Recv: ok N:3831
Send: N3833 G1 X207.302 Y150.613 E41.74077*107
Recv: ok N:3832
Send: N3834 M105*27
Recv: ok N:3833
Recv: ok T:244 /245 B:101 /100 @:61 B@:0
Send: N3835 G1 X207.302 Y151.190 E41.75883*98
Recv: ok N:3835
Send: N3836 G1 X213.909 Y157.797 E42.05163*97
Recv: ok N:3836
Send: N3837 G1 X213.909 Y158.373 E42.06968*97
Recv: ok N:3837
Send: N3838 G1 X207.302 Y151.766 E42.36249*104
Recv: ok N:3838
Send: N3839 G1 X207.302 Y152.342 E42.38054*104
Recv: wait
Send: N3840 G1 X213.909 Y158.949 E42.67335*103
Recv: ok N:3839
Send: N3841 G1 X213.909 Y159.526 E42.69140*108

when it says ´´wait´´ it stops

UnchartedBull, 2021-09-10:
The wait would indicate that there is something funky going on with the USB connection to the printers mainboard and not necessarily the Pi hitting some kind of bottleneck.

Here are the next steps I would propose:
- Remove OctoDash once again and start a print without OctoDash installed
- If the problem continues replace the USB cable
- If that doesn't fix it try and update your Marlin / Klipper Version
- If all above fails try with a print from an SD Card and check if the issue still occurs.

GN1G, 2021-09-11:
> The wait would indicate that there is something funky going on with the USB connection to the printers mainboard and not necessarily the Pi hitting some kind of bottleneck.
>
> Here are the next steps [...]

After I removed OctoDash the problem was still there, so I printed directly from the USB drive and everything worked normal.
I set up a new OctoPi and everything worked fine again. Will try to install OctoDash again on Monday.

Is it possible, that the Qidi Firmware is a bit of a problem? because their Firmware is closed source, but Marlin based.

UnchartedBull, 2021-09-15:
The Qidi firmware shouldn't be an issue, I'd rather suspect that during an update (could be OctoPrint, System or OctoDash) something got mixed up that caused the issue. Please let me know if the issue occurs again.

closed 2021-10-12 by stale[bot]