UnchartedBull/OctoDash #2278: Custom Action [Web-Post]

SaKiEQ, 2021-09-21:
Me again.... 

After several different setups and different problems to solve in creative ways, the simple and dumb issue of controlling a printer / cabinet light came back to me. 

This time 'Action Commands' won't work, since the printer firmware does not deal with M118 or M117 for its chinese wisdom... 
Triggering GPIOs as on and off, is one thing, while I rather use a python script that toggles, rather then has two discrete states and therefore two buttons/commands. 

While I can use Enclosure plugin, to trigger shell script, eg. python script, the only way of doing so is by [!WEB] action command, but then that opens a web window every time (darn). 

A dedicated action command [!WEB-POST] could be a solution, as any link could be pushed/posted without window interaction, just send and execute the link. Where Enclosure Plugin can do its thing at the other end. 

There are many cases where I could use this, all my printers have now WS-Leds on the print head, for illumination and status indicators, but the are all running from separate MCUs, that can either receive MQTT or Web interactions. 

(I would change the firmware of the printer, if it would be documented... thanks to our friends overseas for copying Marlin but doing a poor job at it). 

Thanks !

UnchartedBull, 2021-09-22:
That should be fairly easy to do. It will be a simple POST Call with no Body or Authorization, I don't know whether this will help with your problem.

Also the Enclosure output will have a custom action in the next release which will enable you to trigger outputs without calling web :) It's already in the wiki but not released yet: https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions#predefined-actions

SaKiEQ, 2021-09-22:
Hah... I was trying the [!OUTPUT] yesterday, after discovering it in the wiki and could not get it to work - now i know why ! 

One thing that would help, if the status of the call would not be limited to true/false on/off but also a logical NOT function, to achieve toggling. I am doing that right now via a python script:

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(6, GPIO.OUT) 

state = GPIO.input(6)

try:
        if state == True:
                GPIO.output(6,GPIO.LOW)
                print ("Light ON")
        else:
                GPIO.output(6,GPIO.HIGH)
                print ("Light OFF")

except:
        GPIO.cleanup()

Which is a bit of 'hack' since I am reading the port, despite being defined as output - but it works. 

Since the buttons on the touch screen are precious, having two (one - on, one - off) is a waste, simple toggle would be most helpful. But then thats a Enclosure Plugin feature, not OctoDash.

UnchartedBull, 2021-09-22:
Yes, the problem is that the enclosure plugin still hasn't been upgraded to Python 3, so I might need to look into alternatives. If someone knows something please let me know.

Also with the latest release you should be able to create an infinite amount of custom actions (need to be done via manually editing the custom actions or via jneilllis configuration tool).