UnchartedBull/OctoDash #2308: Detect missing desktop environment and offer to install one when setting up

blame-lewis, 2021-10-01:
**Is your feature request related to a problem? Please describe.**
Yes. When setting up OctoDash, the frontend won't function unless a desktop environment is present, and a stock OctoPi install does not have one. The user may or may not realise that installing one is necessary or know how to.

**Describe the solution you'd like**
1. OctoDash's installation script should check whether a desktop environment is installed.
2. If one is not installed, it should prompt the user with a statement like "OctoDash can't work without a desktop environment. Would you like to install the stock Raspberry Pi one?"
3. If the user selects yes, it's installed - `apt-get -y dist-upgrade && apt-get install raspberrypi-ui-mods` is fine.

**Link other projects**
Nothing related

**Additional context**
Nothing related

UnchartedBull, 2021-10-01:
OctoDash will install ratpoison as a Window manager, which is sufficient. There is no need to install the full-blown raspbian desktop environment, as this will just slow down the boot. I've been running OctoDash since the start now without any desktop environment at all.

Did you run into any issues while installing?

blame-lewis, 2021-10-02:
Hi Timon,
I did run into issues, yes. What I did was:

   1. Take the stock OctoPi image
   2. Install OctoDash as per instructions
   3. Install Waveshare LCD-show
   4. Restart

What happened was X refused to boot, stating there was no screen attached.
Installing the Raspbian desktop environment solved this. It still booted
into ratpoison and didn't launch the full desktop environment, so I'm
guessing there's a missing dependency in there somewhere.
Thanks
Lewis

UnchartedBull, 2021-10-09:
Ah ok, I think the issue here is that you installed the drivers after the OctoDash installation, which probably messed something up. The driver installation scripts usually do some major changes to the files that OctoDash is also changing to automatically start up. I'm closing this here for now, since the full Desktop environment isn't really the ideal solution (way overkill for a single window). If you'd like to tinker with this feel free to execute your steps on a fresh SD Card and swap step 2 & 3 and report back.

Thanks for the info though!