UnchartedBull/OctoDash #2357: Install on Raspberry 4 ?

3DFans, 2021-10-17:
When I try to install it, I get the following error. Can anyone help me with this? 

`pi@octopi:~ $ bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)
Installing OctoDash v2.2.0, armv7l
Installing Dependencies ...
25 packages can be upgraded. Run 'apt list --upgradable' to see them.
libavahi-compat-libdnssd1 is already the newest version (0.7-4+deb10u1).
libuuid1 is already the newest version (2.33.1-0.1).
libxss1 is already the newest version (1:1.2.3-1).
libxss1 set to manually installed.
libxtst6 is already the newest version (2:1.2.3-1).
libxtst6 set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libappindicator3-1 : Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
                      Depends: libdbusmenu-gtk3-4 (>= 0.4.2) but it is not going to be installed
                      Depends: libindicator3-7 but it is not going to be installed
 libgtk-3-0 : Depends: adwaita-icon-theme but it is not going to be installed
              Depends: libatk1.0-0 (>= 2.35.1) but 2.30.0-2 is to be installed
              Depends: libc6 (>= 2.29) but 2.28-10+rpi1 is to be installed
              Depends: libcairo-gobject2 (>= 1.14.0) but it is not going to be installed
              Depends: libgdk-pixbuf-2.0-0 (>= 2.40.0) but it is not installable
              Depends: libglib2.0-0 (>= 2.59.0) but 2.58.3-2+deb10u2 is to be installed
              Depends: libjson-glib-1.0-0 (>= 1.5.2) but it is not going to be installed
              Depends: libpango-1.0-0 (>= 1.45.5) but 1.42.4-8~deb10u1 is to be installed
              Depends: libxcomposite1 (>= 1:0.4.5) but 1:0.4.4-2 is to be installed
              Recommends: libgtk-3-bin but it is not going to be installed
              Recommends: librsvg2-common but it is not going to be installed
 xserver-xorg : Depends: xserver-xorg-core (>= 2:1.17.2-2) but it is not going to be installed
                Depends: xserver-xorg-video-all but it is not going to be installed or
                         xorg-driver-video
                Depends: xserver-xorg-input-all but it is not going to be installed or
                         xorg-driver-input
E: Unable to correct problems, you have held broken packages.

Couldn't install dependenices!
Seems like there is something wrong with the package manager 'apt'

If the error is similar to: 'E: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'stable' to 'oldstable''
you can run 'sudo apt update --allow-releaseinfo-change' and then execute the OctoDash installation command again
pi@octopi:~ $
`

Jack77777777, 2021-10-17:
Hey man,

The answer is in the output, isn't it ?

3DFans, 2021-10-17:
> Hey man,
> 
> The answer is in the output, isn't it ?

`pi@octopi:~ $ sudo apt update --allow-releaseinfo-change
Hit:1 https://archive.raspberrypi.org/debian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
13 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list.d/raspi.list:4
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list.d/raspi.list:5
pi@octopi:~ $
`

:D

UnchartedBull, 2021-11-03:
Something is majorly messed up with your package manager. Closing this here as this isn't an issue with OctoDash really.

closed 2021-11-03 by UnchartedBull