UnchartedBull/OctoDash #2391: Forcing the PI to do Https breaks the connection

str3den, 2021-10-31:
**Describe the bug**
Hi im doing selfsigned signature in octopi and it't breaks the connection to octodash, I tryed editing the config file but it seams there is no support for https??

UnchartedBull, 2021-11-03:
I don't see an issue with using https. Have you tried editing the config manually?

str3den, 2021-11-03:
Yes edited the config file manually with Https:// but could not complet the opstart guide.

And once I removed the forced Https on the PI everything worked fine.

Sendt fra Mail<https://go.microsoft.com/fwlink/?LinkId=550986> til Windows

UnchartedBull, 2021-11-08:
Yes you probably need to create the config completely manually, since the wizard is not setup to accept `https`. So you only can change to `https` after the installation wizard or create the config manually from the ground up.

str3den, 2021-11-08:
Yea sorry if that wasn't clear, also tryed after the guide completed edit the config file to https the program simply wont accept a adresse with https://

UnchartedBull, 2021-11-10:
Oh that is interesting. Going to test tomorrow and see how to fix that.