UnchartedBull/OctoDash #2402: PSU Control broken on OctoPrint 1.7.2 update

SaKiEQ, 2021-11-04:
**Describe the bug**
When waking up Octodash by touching the screen, octoprint server turns the printer on then off, or if already running toggles the printer off instead of waking  up. 

Everything was working fine just prior the Octoprint update. From within octoprint everything regarding PSU control seems to work just fine. 

Having said that, there is other things within Octoprint that are doggy in my setup after update, like remote hosted print files (via file mount) are not working, maybe that error trips OctoDash on startup.

UnchartedBull, 2021-11-18:
Mhh that is really weird, OctoDash should only send a single command to OctoPrint and not multiple. Could you maybe check the OctoPrint log if there is something useful in there? Thanks!

SaKiEQ, 2021-11-18:
Hi there, I deleted config and removed / reinstalled PSU plugin, since then it has been working again. Not sure what was the cause of the issue after updating Ocotprint. 

There are still issues at time where OctoDash has a really difficult time waking up and reconnecting with OP, then other times not at all - but that is a separate issue all together.

closed 2021-11-18 by SaKiEQ

UnchartedBull, 2021-11-18:
Please feel free to open a new issue for that if you feel like it :). Also great that the Problem figured itself out, thanks for the update!

SaKiEQ, 2021-11-18:
Could have been a caching / reboot issue... this was a major update on my system, since I went from Python2 -> Python3 in the same time... so stuff is bound to bugger up :)

Maybe updating plugins, OctoDash and OctoPrint at the same time with major releases is not ideal...

UnchartedBull, 2021-11-18:
Yes multiple upgrades are always bound to cause issues. But I'm doing the same, so I know the struggle to find out what actually went wrong.

SaKiEQ, 2021-11-18:
I wait and wait and wait to update, since my printer(s) are usually quiet busy... so when I have an afternoon of light and space, I update the whole enchilada that piled up - followed by much longer debugging. 

But I have to say there are no moments where I learn more about Raspeberrys, Linux and Python than during debugging !!