UnchartedBull/OctoDash #2412: RobotDigg 5" LCD Screen XPT2046 Support (Working)

gordo3di, 2021-11-10:
**General Information:**
 - Using Raspberry Pi 3B+
 - RobotDigg 5" LCD Touchscreen XPT2046

**Additional context**
For anyone who wants to use this screen you need to make the following changes in the config for the touchscreen input to work. Just posting here to help anyone out.

https://forums.raspberrypi.com/viewtopic.php?t=143581

```
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=ads7846,penirq=25,speed=10000,penirq_pull=2,xohms=150
```

scottmudge, 2021-12-09:
This isn't really an issue related to OctoDash. OctoDash doesn't necessarily care what kind of display is attached... in theory any screen attached to a RaspberryPi should. Of which there are dozens of tiny touchscreens.

gordo3di, 2021-12-09:
> 

Gotcha. It wasn't an issue, I was just posting my config that worked to help anyone in the future.

closed 2021-12-09 by gordo3di