UnchartedBull/OctoDash #2436: Install script is not working anymore

roelbbzz, 2021-11-16:
Tried to install octodash with the script provided in the wiki. It gave some errors, so I tried manually.
errors in script 

 ```
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/libx/libx11/libx11-dev_1.6.7-1+deb10u1_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rxvt-unicode/rxvt-unicode_9.22-6_armhf.deb  404  Not Found [IP: 93.93.128.193 80]
E: Failed to fetch http://raspbian.raspberrypi.org/raspbian/pool/main/r/rxvt-unicode/rxvt_2.7.10-7.1+urxvt9.22-6_all.deb  404  Not Found [IP: 93.93.128.193 80]
```

All is going well until I want to install 

` sudo apt install xserver-xorg ratpoison x11-xserver-utils xinit libgtk-3-0`
then I get the same errors. 

Where can I get the files needed for the install. 

Thanks in advance

jneilliii, 2021-11-16:
try `sudo apt update` first?

roelbbzz, 2021-11-17:
> try `sudo apt update` first?

Thanks, that worked

closed 2021-11-17 by roelbbzz