UnchartedBull/OctoDash #2452: Touchinterface is rotated 90 degrees

lpvezina, 2021-11-19:
Octodash is installed, ts_calibrate and ts_test worked perfectly.  When opening Octodash, the screen is rotated 90 degress.  I've installed waveshare and goodtft drivers.  I saw in the troubleshooting section the following solution: Install lightdm through apt, run the installer script again, and then remove lightdm. Reboot.  I did it but with no luck, screen is still rotated.  

I also tried the following:
- Added _display_rotate=3_ to _/boot/config.txt_

Can you help me to resolve this issue?

jneilliii, 2021-11-19:
You may find this helpful, it's what worked with my hyperpixel.

https://community.octoprint.org/t/inverting-an-hdmi-display-on-the-raspberry-pi-4b/22804?u=jneilliii

UnchartedBull, 2021-11-29:
Please reopen if the linked fix doesn't solve your issue.

closed 2021-11-29 by UnchartedBull