UnchartedBull/OctoDash #2467: Display stays black after boot

slewf0ot, 2021-12-01:
After booting my RPi 4 (Bullseye) Octoprint Version 1.7.2 I can see the screen do everything it is supposed to, ratpoison comes up, but then the screen just goes black. I've tried doing the export DISPLAY=:0 command followed by octodash and all I get is this...

selected language: en
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
[1794:1201/070238.661329:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap

Any suggestions?

slewf0ot, 2021-12-01:
found if I launch octodash like this..
`octodash --disable-seccomp-filter-sandbox`
everything comes up fine and works.

closed 2021-12-01 by slewf0ot

wohali, 2022-01-13:
FYI for those looking to get this working under `bullseye` (latest RPi OS release as of Nov 2021), this is the fix you need. Edit the `octodash` line in `/home/pi/.xinitrc` and add ` --disable-seccomp-filter-sandbox` to the end. Reboot.