UnchartedBull/OctoDash #2485: PSUControl though screen with Pi Power Button on GPIO3

arut16, 2021-12-04:
**Describe the bug**
Hi all,

I'm using an RPi4 with an Octopi/Octoprint instance on it for 3D printer. In Octoprint software, i'm using
OctoPrint-PSUControl plugin to control a 3v3 relay on GPIO21 and through it power on and down 3D printer with Raspberry. And without the physical power button, this working as it must.

I successfully installed the power-button on GPIO3 from **[this repo](https://github.com/Howchoo/pi-power-button)** and powering on and shutdown of raspberry is working well through the momentary button.

BUT. Then I touch the screen to power on the printer through PSUControl. The relay closes and the printer power on and at the same time raspberry seems to "receive" also the command to power off like if I push the new power button but I didn't touch it !

**This behavior does not happen then I power on and off printer though PSUControl button on Octoprint webinterface.**

There must be a kind of interference of signal between [Pi Power Button](https://github.com/Howchoo/pi-power-button) and OctoPrint-PSUControl then I'm trying to power on printer by touching the OctoDash screen. But I don't know where to start in investigations...

Any help will be very kind !

**To Reproduce**
Steps to reproduce the behavior:
1. Install [Pi Power Button Repo](https://github.com/Howchoo/pi-power-button) and a momentary button on GPIO3 and GND.
2. Power On RPi and wait to octoprint startup.
2. Activate function "Turn On Printer When Exiting Sleep" [by touching the screen] in OctoDash settings.
3. Touch the screen.

**Expected behavior**
Printer must power on through the relay AND raspberry stay ON.

**General Information:**
 - Hardware : Raspberry Pi 4 2Go, a 3v relay connected to GPIO21, a momentary button connected to GPIO3(SCL) and GND, a 5" capacitive HDMI screen (MPI5001) 800x480 USB touch interface
 - OS Info : OctoPi 0.18.0
 - OctoDash Version : 2.2.0
 - OctoPrint Version : 1.7.2
 - Python version : 3.7.3

**Systeminfo Bundle**

[octoprint-systeminfo-20211204191905.zip](https://github.com/UnchartedBull/OctoDash/files/7654505/octoprint-systeminfo-20211204191905.zip)

**Octoprint.log**
[octoprint.log](https://github.com/UnchartedBull/OctoDash/files/7654507/octoprint.log)

jneilliii, 2021-12-04:
Probably a PSU Control plugin related issue rather than an octodash issue. OctoDash just sends the API command to PSU Control to power on I'm pretty sure.

arut16, 2021-12-04:
> Probably a PSU Control plugin related issue rather than an octodash issue. OctoDash just sends the API command to PSU Control to power on I'm pretty sure.

You maybe right... Then I toggle psu button or the physical button, I see this lines :
[image]
Do you understand something ?
Can this issue be from this "module g2d_23 not found in directory /lib/modules/5.10.63-v7" ?

UnchartedBull, 2022-01-17:
Please google around for this error. As this is not related to OctoDash I'm closing this here for now.

closed 2022-01-17 by UnchartedBull