UnchartedBull/OctoDash #2498: M600 not prompting

pennystockss, 2021-12-12:
**What doesn't work?**
Sending M600 via Octoprint or through Filament Change does not prompt for continue/purge more.
Filament change from GUI just says M600 sent, follow instructions on printer, I'll wait for you.

**What did you already try?**
Host action commands are enabled by default it would seem. Octoprint on the web interface will give me popups about it and I can complete a M600 via Octoprint web interface. Octodash there are no prompts, even for the M600 filament change from the GUI.
Also tried disabling plugins for Host commands in OctoPrint to see if OctoDash can see them.

**General Information:**
 - Hardware Raspberry Pi 3 with stock BIQU BX
 - OS Info OctoPi 18
 - OctoDash Version 2.2.0
 - OctoPrint Version 17.2

**Additional context**
I'm not sure honestly if OctoDash supports action host commands like OctoPrint or if it is OctoPrint intercepting it so OctoDash can't see it. I see M600 mentioned through the changelogs so I assumed it's supported. Sorry I'm a newb!

With the BIQU BX I can toggle back to the Marlin LCD emulator but if I do that, it gets stuck in a loop where the M600 repeats indefinitely and keeps trying to change filament over and over as you complete them. Then when I go back to OctoDash, the printer freezes, loops the last few bits of gcode then eventually throws a M112. This only happens during a print, I can toggle back and forth at idle.

jneilliii, 2021-12-12:
OctoDash doesn't support host action prompts. There was a PR from @pciavald that they closed for some reason [here](https://github.com/UnchartedBull/OctoDash/pull/1661). I do think it would be a good addition to the platform.

UnchartedBull, 2022-01-17:
Will be hopefully included in the next release. Duplicate of #1624.

closed 2022-01-17 by UnchartedBull