UnchartedBull/OctoDash #2499: Display Layer Progress - Remove dependencies

synman, 2021-12-12:
**Is your feature request related to a problem? Please describe.**
Display Layer Progress isn't for everyone.  It has been suggested to remove it from the dependencies list previously.  It's also stated on the OctoDash homepage this is something being considered but I was unable to locate an actual issue tracking its progress.

**Describe the solution you'd like**
Integrate dependent APIs into your own plug-in, call it the "OctoDash Plugin", or integrate the functionality directly into OctoDash.

**Link other projects**
N/A

**Additional context**
This may not be for everyone and may not align to your roadmap.

jneilliii, 2021-12-12:
current stable release actually doesn't require it, even though the installer script I think still flags it as mandatory.

synman, 2021-12-12:
Would love to just uninstall it

jneilliii, 2021-12-12:
You can from octoprint, and then disable the option in octodash settings.

synman, 2021-12-13:
I pulled down the latest octodash binary last night, installed it, and noticed that I could not disable the DLP plugin.  Does this behavior change if you yank DLP out from under OctoDash?

jneilliii, 2021-12-13:
I run without the plugin installed without errors. You may need to disable the option in config.json manually, or use my octodash companion plugin to disable and then reboot the pi.

synman, 2021-12-13:
ok... I got enough here to do the deed.  thanks for all the info

UnchartedBull, 2022-01-17:
https://github.com/UnchartedBull/OctoDash/pull/2580 also removed it from the install script. It should also be editable via the settings GUI now.

closed 2022-01-17 by UnchartedBull