UnchartedBull/OctoDash #2531: Octodash not working on RPI OS Bullseye

supawiz6991, 2021-12-19:
**What doesn't work?**
Octodash fails to run on RPI OS Bullseye, result its either a blank screen before crashing or it crashes immediately. 

**What did you already try?**
I tried installing octodash via the automatic script. When that failed I tried the manual method. Manual method could not be completed as it requires the package gir1.2-gnomekeyring-1.0 which was deprecated in Buster and removed completely in Bullseye. I did attempt continuing on without it but Octodash wouldn't run.

**General Information:**
 - Rasperry Pi 4B 4GB, 7' Raspberry pi Touch Display
 - RPI OS Bullseye (Did not use Octopi as octopi isn't configured with Bullseye).
 - OctoDash Version [e.g. v2.2.0]
 - OctoPrint Version [e.g. v1.7.2]

**Additional context**

UnchartedBull, 2022-01-17:
That's interesting. I'll look into that soon.

brad07x, 2022-01-31:
Think I may have stumbled onto this as well, albeit with a slightly different presentation for the error message.

- Raspberry Pi 4B 4GB, 7" Official RPI Touch Display (DSI)
- CustomPiOS build for OctoPi image against RPI OS Bullseye
- OctoDash Version: v2.2.0 (via install script)
- OctoPrint version: 1.7.3
- OctoPi version: 1.0.0 (custom build)

Used OctoPi's [custom/dev build instructions](https://github.com/guysoft/OctoPi/blob/devel/README.rst#Developing) to roll my own updated OctoPi image with RPI OS Bullseye as the source distribution.

Install script encountered an error with installing the python package RPi.GPIO, but continued.
```
      /usr/bin/ld: build/temp.linux-armv7l-3.9/source/soft_pwm.o:/tmp/pip-install-e9pn5v9u/rpi-gpio_ca6b738aa3de4aacb73c95858be6fa94/source/soft_pwm.c:28: multiple definition of `threads'; build/temp.linux-armv7l-3.9/source/event_gpio.o:/tmp/pip-install-e9pn5v9u/rpi-gpio_ca6b738aa3de4aacb73c95858be6fa94/source/event_gpio.c:60: first defined here
      collect2: error: ld returned 1 exit status
      error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> RPi.GPIO
```

Boots to console autologin, shows ratpoision welcome banner, a grey screen, and then X crashes.

```
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
... FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
```

Tried disabling GPU Acceleration by commenting the `dtoverlay=vc4-kms-v3d` in /boot/config.txt.

After doing this, the screen had less artifacts (no brightness setting oddities, and no glitched characters in the text console). Octodash also appeared to start again after the first crash, but both attempts ultimately resulted a crash with the error described above.

Trying to run octodash manually also results in a segfault, just like @supawiz6991.
```
pi@octopi-cr30:~ $ /usr/bin/octodash
selected language: en
Segmentation fault
pi@octopi-cr30:~ $ /usr/bin/octodash --no-sandbox
selected language: en
Segmentation fault
pi@octopi-cr30:~ $ sudo /usr/bin/octodash --no-sandbox
[sudo] password for pi:
selected language: en
Segmentation fault
pi@octopi-cr30:~ $
```

jalanjarosz, 2022-01-31:
Out of curiosity, do you have package libappindicator3-1 installed?

This sounds very similar to my issue in Armbian Bullseye. Raspbian Bullseye and Armbian Bullseye are both Debian 11 sources.  

Check issue #2589 for my issue.  End results are the same; video crashing when starting.

brad07x, 2022-01-31:
@jalanjarosz 

Yes, that package is present:
```
pi@octopi-cr30:~ $ sudo dpkg -l | grep libappindicator
ii  libappindicator3-1:armhf             0.4.92-8                         armhf        allow applications to export a menu into the panel -- GTK3 version
```

brad07x, 2022-01-31:
Update - **I was able to get OctoDash running!**

Resolution on my system was:

1. Install DE with  `sudo /home/pi/scripts/install-desktop`, answering 'Yes' to start DE at boot at the end of the packages installation.
2. Reboot 
3. DE started successfully and didn't crash - progress!
4. Use the menu to start octodash with the `Run...` window, or the graphical terminal XVT (`System Tools -> Xvt`).
5. Using either of these methods, the invocation that worked was `/usr/bin/octodash --no-sandbox`. In my setup, this starts octodash running under the unprivileged `pi` user. Omitting the `--no-sandbox` option resulted in a crash and the same seccomp-bpf errors I described earlier.

An educated guess - it seems like the changes to the system from installing the full DE made the difference. I had already tried running `/usr/bin/octodash --no-sandbox` as `pi` when the system was previously set up by the octodash install script with ratpoison as the lightweight WM.

Due to the similarity with the error messages described in this [Arch Linux ARM chromium forum topic](https://archlinuxarm.org/forum/viewtopic.php?f=15&t=14453) and this [Vivaldi forum topic](https://forum.vivaldi.net/topic/53399/crashing-seccomp-bpf-failure-in-syscall-0403-rpi4-8gb-ubuntu20-10-arm64-armhf), I figured it would be worth disabling GPU acceleration (as described earlier) and trying the `--no-sandbox` flag with Octodash.

After the install of the full DE, the system is configured with:

- DM: LightDM
- DE: LXDE
- WM: mutter

I attempted to get things running with GPU Acceleration enabled again (uncomment `dtoverlay=vc4-kms-v3d`) in /boot/config.txt, but LXDE is glitchy/non-working on the system in this state. Touch & Mouse work, but interactions with the LXDE UI menus and right-clicks don't work at all, or not reliably/as expected. Windows are partially drawn (i.e. outline & shadow drawn, but no content) and behavior is generally very glitchy. 

I'm going to continue testing with GPU Acceleration off on my Pi since it doesn't seem to perform poorly with it turned off/like it would even be needed. I also plan to build an up-to-date CustomPiOS OctoPi image with raspbian/buster as the source to see if that yields any improvements.

Here is some additional detail on package versions from the system:
```
Script started on 2022-01-31 17:44:21-05:00 [TERM="xterm-r6" TTY="/dev/pts/0" COLUMNS="80" LINES="24"]

------------------------------------------------------------------------------
Access OctoPrint from a web browser on your network by navigating to any of:

    http://octopi-cr30.local
    http://<ip>

https is also available, with a self-signed certificate.
------------------------------------------------------------------------------
OctoPrint version : 1.7.3
OctoPi version    : 1.0.0
------------------------------------------------------------------------------

### Terminal logins still try to start octodash. Break with Ctrl+C
/usr/lib/xorg/Xorg.wrap: Only console users are allowed to run the X server
^Cxinit: unexpected signal 2
X connection to :0 broken (explicit kill or server shutdown).

### Terminal logins still try to start octodash - for a second time. Break with Ctrl+C.
/usr/lib/xorg/Xorg.wrap: Only console users are allowed to run the X server
^Cxinit: unexpected signal 2
X connection to :0 broken (explicit kill or server shutdown).

### Manual run w/o --no-sandbox flag.
pi@octopi-cr30:~$ /usr/bin/octodash
selected language: en
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
../../sandbox/linux/seccomp-bpf-helpers/sigsys_handlers.cc:**CRASHING**:seccomp-bpf failure in syscall 0403
[2703:0131/174450.085697:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap

### Manual run w/ --no-sandbox flag - it worked!
pi@octopi-cr30:~$ /usr/bin/octodash --no-sandbox
selected language: en
pi@octopi-cr30:~$

pi@octopi-cr30:~$ echo "Successful Run!"
Successful Run!

[... 45 lines cut ...]

pi@octopi-cr30:~$ dpkg -l | grep lightdm
ii  liblightdm-gobject-1-0:armhf         1.26.0-7+rpt1                    armhf        simple display manager (GObject library)
ii  lightdm                              1.26.0-7+rpt1                    armhf        simple display manager
ii  lightdm-gtk-greeter                  2.0.8-2                          armhf        simple display manager (GTK+ greeter)

### OS Info
pi@octopi-cr30:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

pi@octopi-cr30:~$ uname -a
Linux octopi-cr30 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux

pi@octopi-cr30:~$ date
Mon 31 Jan 17:48:52 EST 2022

pi@octopi-cr30:~$ exit
exit

Script done on 2022-01-31 17:48:55-05:00 [COMMAND_EXIT_CODE="0"]
```

supawiz6991, 2022-02-11:
> Install script encountered an error with installing the python package RPi.GPIO, but continued.
@brad07x 
The RPI.GPIO install failure is a known issue. The dev of this package fixed the problem in alpha version 0.7.1a4.

According to the ticket on the devs page (https://sourceforge.net/p/raspberry-gpio-python/tickets/198/), full release 0.7.1 has been released (5 days ago) and should contain the fix. That being said, I've not yet tested to see if that is the case.

pip index versions rpi.gpio does show version 0.7.1 is available and thus the RPI.gpio error should no longer occur as long as the fix is in there as the dev specified.

Should this not be the case, the work around to use the alpha version is below. ONLY USE THIS IF THE FULL RELEASE DOESN"T WORK.
The work around is generally applied during octoprint setup:

source ~/oprint/bin/activate

then

pip install RPi.GPIO==0.7.1a4

I'll spin up a fresh instance of Bullseye, sometime in the next couple of days and see if the RPI.gpio error goes away and report back.

supawiz6991, 2022-02-12:
I tested this out and RPi.GPIO 0.7.1 does include the fix from the alpha, so the work around is no longer needed.

In regards to libayatana-appindicator3-1: I'm getting the error:

Package 'libappindicator3-1' has no installation candidate

It would appear that the package has been renamed libayatana-appindicator3-1.

I modified the install.sh script with the updated package name and was able to get  through the installation process without issue.

This confirms that RPi.GPIO 0.7.1 resolved the error where it couldn't be installed.

This brings us back to the original Segmentation fault.

supawiz6991, 2022-05-23:
Hi,
Just checking in to see if there is any update on this and/or an eta on a fix. 
Thanks!

UnchartedBull, 2022-05-23:
Hi, sorry for not responding earlier. I think the best option here would be to just try and install OctoDash on a clean OctoPi installation (not using the script, just executing `sudo dpkg -i octodash.deb`. That should then list the correct dependencies that are missing, which can the be added to the install script.

I have to search for a Pi that isn't in use at the moment, since it's impossible to buy a new one atm. So might be a couple of days.

supawiz6991, 2022-05-24:
No worries @UnchartedBull . I feel your pain on the availability of pi's.

So, I went to test this out again and discovered a critical changing on the octopi's end. When I opened this ticket there was no stable 64-bit version of octopi. There was a nightly but it was not stable.

When I tested this I wrote Raspberry pi OS Lite to my memory card and setup octoprint manually.

Today I checked in with the Octopi github and there has been significant changes in the progress of a 64-bit build.

First, Octopi has pivoted to an ubuntu base for their 64-bit build due to issues with Raspberry Pi OS and their inconsistent release schedule.

I did retest under Raspberry pi os Bullseye lite in case you still want to support it going forward for manual installations:

Using your .deb method, 

libappindicator3-1 shows not available candidate. As I found before this package has been renamed libayatana-appindicator3-1.

girl.2-gnomekeyring-1.0 - This was deprecated in Buster and removed completely in Bullseye. 

Results:
I installed the renamed package libayatana-appindicator3-1 and skipped girl.2-gnomekeyring-1.0 and was able to get Octodash to successfully install.

Recommendations: 
1. Verify that libayatana-appindicator3-1 is a legitimate rename of libappindicator3-1 and then update the install script accordingly.
2. In regards to girl.2-gnomekeyring-1.0; I don't know what this does for octodash...but if its important then a replacement will need to be found and the script updated.

At a minimum this should get people working on Raspberry pi OS Bullseye.

I'll open a new ticket regarding support for the Ubuntu based image.

facastagnini, 2022-12-14:
Many months later... I ran the installation script and experience the issue.
All I had to do to get octodash to start was adding "--no-sandbox" like this

```

greg10810, 2023-02-03:
@facastagnini 
Excuse my newbie ignorance, your fix above - which file is this in and does the desktop UI need to be installed?

jneilliii, 2023-02-28:
@UnchartedBull with OctoPi 1.0.0 released now with base Bullseye image it may be a good idea to update this.

berb6, 2023-02-28:
@UnchartedBull  I second the above, I am having a tough time setting up OctoDash following #2531 and Jneilliiis help

editwentyone, 2023-10-15:
@jneilliii @UnchartedBull 

i have good news, I got it running on latest `octopi 1.0.0` with `octoprint 1.9.3` build
here is what I did (I think):

1. fresh image build `2023.10.09.154319` from 2023-10-09
OctoPi 1.0.0 with OctoPrint 1.9.3
`https://github.com/OctoPrint/OctoPi-UpToDate/releases/download/1.0.0-1.9.3-20231009154319/octopi-1.0.0-1.9.3-20231009154319.zip`
(that revived my v2 camera with the new camera stack and also if you want to get a waveshare screen running: `https://github.com/goodtft/LCD-show`)

2. Install the Desktop Environment, I did this on my `Pi Zero 2`, no problems so far
```
sudo /home/pi/scripts/install-desktop
Yes
sudo reboot
```

3. `sudo apt install libappindicator3-1`

4. Edit `nano /home/pi/.xinitrc`
```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

#ratpoison&
octodash --disable-seccomp-filter-sandbox

#octodash --no-sandbox &
#exec mutter
```

5. Edit `nano ~/.bashrc` with this at the end
```
source /home/pi/scripts/welcome
if [ -z "$SSH_CLIENT" ] || [ -z "$SSH_TTY" ]; then
   #xinit -- -nocursor
   startx -- -nocursor
fi
```

maybe you can have a look and optimize further? disabled ratpoison and added --disable-seccomp-filter-sandbox, don't know what I am doing but its working :)

Hillshum, 2025-03-21:
OctoDash 2.4.0 now suppports Bullseye.

closed 2025-03-21 by Hillshum