UnchartedBull/OctoDash #2532: Closing Ocotdash screen?

Trekinfo, 2021-12-19:
Hey,

is there a way to close Octodash Screen to get back to the normal Raspberry X-Window?

I tried CTRL-Q but this got me back to the Console and with startx Octodash loads.

Greets Trekinfo

jneilliii, 2021-12-19:
CTRL+T followed by CTRL+K I think. or vice versa.

jneilliii, 2021-12-19:
The OctoDash Companion plugin also has a `Stop OctoDash` option it adds to the system menu of OctoPrint.

Trekinfo, 2022-01-02:
> The OctoDash Companion plugin also has a `Stop OctoDash` option it adds to the system menu of OctoPrint.

Hi,

this works, but this drags me down to the Terminal and with startx i get the Octodash Screen an not the nomal X Window.
I use the Raspyberry also for Pihole, Octoprint, Teamspeak Server and a few other Allpications.
Therefore i need to swap beetween Octodash and the normal X Window for Maintanance.

Greets Trekinfo

UnchartedBull, 2022-01-17:
If you installed Ratpoison there is no normal X Window anymore. The console is your desktop basically. You can launch stuff from there, you won't get any icons or similar though.

If you want to keep your normal desktop environment you should manually install OctoDash (and skip the ratpoison part).

closed 2022-01-17 by UnchartedBull