UnchartedBull/OctoDash #2545: The "Home" button under control

fizzyade, 2021-12-28:
I found out, purely by accident that pressing the Home button under control while printing is a really bad idea.

Occasionally, Octodash doesn't automatically show the print status, it remains on the general information page, although it does show "printing..." in the bottom right.  I know you can flip to the other screen, but for the life of me yesterday, I couldn't remember how.

And this is where things went quickly south with my 3+ hour print.

While trying to find how to flip to the other screen, I went through clicking on various icons on my screen, I went into control and saw the "Home" icon and thought that would take me back to the Home Screen.

What a bad choice that was.

Octodash, without warning or asking me "Do you really wan't to Home the printer while printing, this is a really stupid idea and is going to end up with tears?" promptly stopped the print, moved the head to the bottom left and started the calibration.

At this point, I have a reasonably sized print in the centre of my bed, so as soon as the printer starts printing I rather rapidly hit the reset button, about 9 minutes from he end of the 3 hour print.

You should not be able to do this while printing, or at the very least, the Octodash should ask whether you really want to do what you are attempting to do because it's bad news, for your hotend, your print, your bed and your printer.

I'm astonished that Octodash allowed this to happen, maybe there's some reason you're allowed to do it, maybe if I'd let it run it could have avoided the parts of the bed that would interfere with the extruder, but it certainly shouldn't do what it did without asking.

I never did find out how to flip back to the right screen as I reset the printer and then spend the next 3 hours swearing.

jneilliii, 2021-12-28:
Those are custom actions and are configurable to include a confirmation prompt. What you describe as the reason you were able to get to those buttons happens to me sometimes as well and the only way I've seen to get them back in sync is restarting OctoDash. I added a restart OctoDash option in the OctoDash Companion plugin in order to have a custom action button that does just that for these instances.

fizzyade, 2021-12-28:
Hey,  thanks for the reply.

Ok, I thought I was going mad with that page not appearing correctly...

I think if it ships with an action like this out of the box, where this bug with the state is entirely possible to come up, and which potentially could damage a users machine, then it needs to warn, I don't even care if it warns because it cannot tell what state the machine is, and therefore warns regardless of whether it's printing or not.

I see this issue where the print page doesn't appear a lot, and that means It's potentially really easy to do this and while it now seems obvious that under the context of it being in a control area, I still managed to do it because a home icon is so synonymous as an action to take you back to a home page. 

A warning along the lines of:

"Home will calibrate your Z axis, the build plate MUST be empty.  Are you sure you want to do this?"

Or something like that.

If it can detect whether the printer is idle or not then it should also probably be "doubled warned" with the first one along the lines of.

"This action will stop the current print, are you sure you want to do this?"

And if yes, then show the warning regarding the build plate.

UnchartedBull, 2022-01-17:
The bug where OctoDash stays on the main screen definitely needs to be fixed. 

But including custom error messages for custom actions is just not possible. There are just too many options to catch here. But it maybe a good idea to add some kind of warning screen if a print job is currently printing (configurable per action of course).