UnchartedBull/OctoDash #2546: Can't retrieve layer progress (similar to #1304)

plymatth, 2021-12-29:
**Describe the bug**
Installed OctoDash several times on a new Raspberry Pi 3B+ with Touchscreen.  Constantly receive the following error card:

Can't retrieve layer progress!

Http failure response for 
http://localhost/plugin/DisplayLayerPorgress/values:
404 NOT FOUND_

**To Reproduce**
Steps to reproduce the behavior:
Consistent error card every 'x' seconds, dependent on refresh interval

**Expected behavior**
I expect this error card to go away.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware Raspberry Pi 3B+
 - OS Info OctoPi install
 - OctoDash Version v2.2.0
 - OctoPrint Version v1.7.2
 - 
**Additional context**
Tried everything listed in Issue #1304, except the new "fork" from jneilliii.  Not knowledgeable enough to know how to use that fork.

Changing update interval from 2000ms to 10000ms delays the refresh of the error card
Currently set the OctoDash to port 80.  Originally kept the default port 5000.  Same error, but address was http://localhost:5000/....
Changed from localhost to IP address.  Same error, but localhost was replaced with IP address.
Tried uninstalling and reinstalling OctoDash
During reinstall, confirmed DisplayLayerProgress was enabled (it was a forced check box)
Confirmed API key twice

jneilliii, 2021-12-29:
I don't think my fork is required. If you don't have DisplayLayerProgress plugin installed you just need to disable it from being enabled in OctoDash's settings. If you do have it installed, make sure that the option for websocket is enabled in that plugin's settings.

plymatth, 2021-12-29:
Thanks!  All this time, I thought OctoDash was installing the plugin..... but the plugin is really an OctoPrint thing.  While I was at it, I also added the FilamentManager and the MMU2S plugins.  I'm good to go, thanks for the help!

closed 2021-12-29 by plymatth