UnchartedBull/OctoDash #2547: Support for M117 messages

adrianstoker, 2021-12-31:
It would be amazing if OctoDash displayed M117 messages, maybe in the bottom right hand corner of the screen where it says 'operational' or 'printing'.

jneilliii, 2021-12-31:
I think there may be plans to include this, but it's very reliant on the capabilities of your printer to report that information (action commands) or a plugin will have to monitor the M117 messages and provide it to the websocket for OctoDash to injest. I would be willing to add that to the Companion plugin if @UnchartedBull wants me to. I know that there have been some recent changes for a notification panel that allows for access to custom actions all the time, and the pop-ups go into it now for API errors, etc.

https://github.com/UnchartedBull/OctoDash/pull/2469

UnchartedBull, 2022-01-17:
The plan for the next release is to include action_command_prompts and OctoPrint notifications into OctoDash. Once the next release is finally out we can have a look on how to integrate M117 :)