UnchartedBull/OctoDash #2548: [Notification Center Version] Always sleeps even while printing

jneilliii, 2022-01-02:
**Describe the bug**
I'm not 100% sure if the issue is related to recent changes for Notification Center or not, but my install seems to put the screen to sleep no matter what, even while a connection is established and during printing. This wouldn't typically be an issue but when you touch the screen to wake it back up it instantly opens the cancel/pause printing action screen rather than just waking up the display. 

**To Reproduce**
Install the notification build from my github action [here](https://github.com/jneilliii/OctoDash/actions/runs/1527977960) start print and wait. Once screen goes to sleep click screen and get print job action screen.

**Expected behavior**
Screen to stay on while printing. 

**Screenshots**
N/A

**General Information:**
 - Pi4 with BTT PiTFT5 V1
 - OctoPi 0.18
 - OctoDash Version [here](https://github.com/jneilliii/OctoDash/actions/runs/1527977960)
 - OctoPrint Version 1.7.2

**Additional context**
I don't think I've modified these commands from the default options or not, but for reference these are the sleep settings.

```
"screenSleepCommand": "xset dpms force standby",
"screenWakeupCommand": "xset s off && xset -dpms && xset s noblank",
```

UnchartedBull, 2022-01-19:
Just checked again and I wasn't able to reproduce this. Also some other members on the discord did mention that they also don't experience this behaviour. Is this still an issue?

jneilliii, 2022-01-19:
Yes, nothing has changed on my end, still will sleep while printing. Were there changes since my build date?