UnchartedBull/OctoDash #2562: Chamber temperature from thermistor without Enclosure plugin.

Akegata, 2022-01-11:
It would be great if we could see the chamber temperature from a chamber thermistor on the home screen without using the Enclosure plugin.

When checking the "Heated Chamber" box in the "Print bed and build volume" tab of the printer profile, the chamber temp shows up in OctoPrint. This is the temp I'd like to have next to the bed temp on the OctoDash home page.

lettore, 2022-08-20:
Yes I agree that shouldn't be necessary to install the enclosure plugin when Octoprint already show the temperature in the UI with the chamber temperature reported with the C prefix.

Ybalrid, 2023-05-18:
I also agree, it should be doable. I am unfamiliar with Octoprint's API, but I am guessing it probably expose the chamber temp in a similar way if it exist.

Currently skimming through OctoDash code trying to figure out how these things are accessed. I never touched any TypeScript (and very little JavaScript) but I guess it should not be too hard to imitate what `temperatureHotendCommand` calls?

Ybalrid, 2023-05-18:
https://docs.octoprint.org/en/master/api/printer.html?highlight=chamber#retrieve-the-current-chamber-state

`GET /api/printer/chamber` should return the number as read by OctoPrint without any plugins, if there's a chamber tempterture

Ybalrid, 2023-05-18:
There's a websocket involved?

Sorry, I am not familiar at all with the architecture of the application 🙂

jneilliii, 2023-05-18:
Yeah, octodash switched from api polling to websocket connection

jneilliii, 2023-05-18:
Looks like it's probably being set here:

https://github.com/UnchartedBull/OctoDash/blob/88f5a898a0ce9474325170ea0c825ea1178d3ef8/src/app/services/socket/socket.octoprint.service.ts#L224-L235

Ybalrid, 2023-05-18:
[image]

"current" message do contain chamber temp!

Added a "chamber" temperature in the model to match, and added the following where you pointed at

```ts
      this.printerStatus.chamber = {
        current: Math.round(message?.current?.temps[0]?.chamber?.actual),
        set: Math.round(message?.current?.temps[0]?.chamber?.target),
        unit: '°C',
      }
```

Now, gotta get this on the webpage being displayed somehow!

I have created this draft, so in case what I am doing is not too broken I could turn this into a PR #3654

Ybalrid, 2023-05-21:
If anybody in this thread is interested, the result of the code in the PR linked above adds the temperature gauge at the bottom of the screen like here 

[image]

Caipitrooper, 2023-05-29:
@Ybalrid, yes I'm very interessted. This is exactly what I'm looking for.
Unfortunately I have no idea how I could implement this. I would have to compile the files, but this is definitely out of my comfort zone. Is there a ready made installer file for this?

Caipitrooper, 2023-06-01:
@Ybalrid 
I tried but there are error over errors coming up. I don't even know if the node.js/npm was correctly installed because there were many warnings.
I guess I have to wait until the change is finally in the official release, or do you have a ready package file for Raspberry?

Ybalrid, 2023-06-01:
> @Ybalrid I tried but there are error over errors coming up. I don't even know if the node.js/npm was correctly installed because there were many warnings. I guess I have to wait until the change is finally in the official release, or do you have a ready package file for Raspberry?

No warranties this works for you:

Assuming you're running 64bit, I have a build. But I only tested it on an *Orange Pi 3 LTS* running Armbain. However, I expect it to also run on Raspberry Pi OS or OctoPi on a Raspberry Pi

https://github.com/Ybalrid/OctoDash/releases/download/chambertemp/octodash-with-standard-chamber-temp.zip

Get this, unzip it into your /home/pi directory on your Pi, then run this command

```
chmod +x /home/pi/linux-arm64-unpacked/octodash
```

Then, edit the /home/pi/.xinitrc file to replace the line that says "octodash" alone to `/home/pi/linux-arm64-unpacked/octodash`

That should look like this 
```
#!/bin/sh

xset s off
xset s noblank
xset -dpms

ratpoison&
/home/pi/linux-arm64-unpacked/octodash
```

Then reboot everything.

To undo the above, you just undo the change you did to .xinitrc, and you delete the folder you unzipped.

Caipitrooper, 2023-06-01:
Unfortunately it doesn't work on my Raspberry 4 with Octopi (exec format error).
Thank you very much for trying to help. Maybe I find somebody else who can try to compile it.

Ybalrid, 2023-06-01:
> Unfortunately it doesn't work on my Raspberry 4 with Octopi (exec format error). Thank you very much for trying to help. Maybe I find somebody else who can try to compile it.

It runs 32bit?

Ybalrid, 2023-06-01:
@Caipitrooper try this https://github.com/Ybalrid/OctoDash/releases/download/chambertemp/linux-armv7l-unpacked.zip

Replace arm64 in every file path mentioned above with armv7l and give it a spin

Caipitrooper, 2023-06-01:
Finally it works. Thank you so much @Ybalrid  👍 🥇 
I hope this will soon find its way into the official release as well, since this is a much better solution than the now none-supported enclosure plugin.