UnchartedBull/OctoDash #2563: Disable sleeping mode on wake up

gatesco, 2022-01-11:
Hi,

that’s my little issue.

I mounted my printer and a rpi 3b+ with 3.5 touchscreen under the same smart plug. 

When I turn on the smart plug, both the printer and raspberry turn on but, after initializing, OctoDash goes automatically into sleeping mode and the printer status on Octoprint is "offline". It seems there’s no way to wake up OctoDash from Octoprint web interface, so I have to go near the printer and tap the rpi touchscreen to wake it up. Then the printer status on Octoprint goes to "online" and I can use it.

I need to have OctoDash active on wake up. Is there the possibility to disable the OctoDash sleeping mode when I turn on the printer and the raspberry?

Thanks in advance!

General Information:
 - Hardware: Raspberry Pi 3b+
 - OctoDash Version: 2.2.0
 - OctoPrint Version: 1.7.2

UnchartedBull, 2022-01-17:
There are 2 options here:

- enable automatic connection in OctoPrint. This will automatically connect OctoPrint to the printer and thus OctoDash will never enter the sleep state.
- disable screen sleep completly.

Please reopen if neither of them work for you.

closed 2022-01-17 by UnchartedBull

MarleyMarley627, 2025-02-02:
How or where  ..........can i disable screen sleep completly  ?????