UnchartedBull/OctoDash #2564: SpoolManager vs Filament Manager

MrRexRuthless, 2022-01-12:
**Is your feature request related to a problem? Please describe.** (No)

**Describe the solution you'd like**
Allow modification/change from Filament Manager to Spool Manager.  It's my understanding that the current plugin, filament manager has been depreciated.  I would like the ability to use Spool Manager instead which I believe is currently being worked on.

**Link other projects** (SpoolManager)
https://github.com/OllisGit/OctoPrint-SpoolManager 

**Additional context**
Sorry if this is not the proper format or all of the required information as this is my first time requesting a feature.

meganssmith, 2022-01-17:
I second this request.

UnchartedBull, 2022-01-17:
Already available. Will be shipped with the next release. If you can't wait you can download a dev build from here: https://github.com/jneilliii/OctoDash/actions/runs/1527977960

closed 2022-01-17 by UnchartedBull